emuflight/EmuFlight #326: No ESC's current sensor reading

dentex, 2020-07-19:
**Describe the bug**
Hello.
I enabled `TELEMETRY` and `ESC_SENSOR` features and the correct serial port. ESC is a Furling32 (BLHeli_32) and the FC is an Airbot F7. I receive the ESC's temperature only. No current reading, enabling the the sensor in the configurator. Is this a known problem? What am I missing?

**To Reproduce**
I enabled `TELEMETRY` and `ESC_SENSOR` features and the correct serial port. I enabled the sensor reading in the configurator.

**Expected behavior**
Current usage in both OSD and telemetry

**Flight controller configuration**
```
# diff all

# version
# EmuFlight / AIRBOTF7 (ABF7) 0.3.1 May 19 2020 / 04:19:02 (9366544) MSP API: 1.47

board_name 
manufacturer_id 
mcu_id 002900313037511839373532
signature 

# reset configuration to default settings
defaults nosave

# name
name EmuAirbotF7

# resources
# ESC ruotato di 90° in senso orario
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 C08
resource MOTOR 4 C09

# mixer

# servo

# servo mix

# feature
feature GPS
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 0 115200 57600 0 115200
serial 1 1024 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led
led 0 7,14::AW:0
led 1 8,14::AW:0
led 2 7,15::AW:0
led 3 8,15::AW:0

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 1 900 1300 0 0
aux 3 26 0 1700 2100 0 0
aux 4 35 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set imuf_roll_q = 4000
set imuf_pitch_q = 4000
set acc_calibration = -42,65,7
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set bat_capacity = 1000
set vbat_max_cell_voltage = 42
set vbat_min_cell_voltage = 35
set vbat_warning_cell_voltage = 36
set current_meter = ESC
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 0
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = ON
set osd_rssi_alarm = 25
set osd_tim1 = 1282
set osd_vbat_pos = 8
set osd_rssi_pos = 2126
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2487
set osd_remaining_time_estimate_pos = 428
set osd_g_force_pos = 2442
set osd_vtx_channel_pos = 225
set osd_current_pos = 2082
set osd_mah_drawn_pos = 2050
set osd_craft_name_pos = 480
set osd_gps_speed_pos = 2529
set osd_gps_lon_pos = 2538
set osd_gps_lat_pos = 2506
set osd_gps_sats_pos = 2497
set osd_home_dir_pos = 2447
set osd_home_dist_pos = 2449
set osd_altitude_pos = 2295
set osd_pid_roll_pos = 256
set osd_pid_pitch_pos = 288
set osd_warnings_pos = 2474
set osd_avg_cell_voltage_pos = 2061
set osd_pit_ang_pos = 193
set osd_disarmed_pos = 2412
set osd_nvario_pos = 248
set osd_esc_tmp_pos = 2552
set osd_esc_rpm_pos = 65
set osd_core_temp_pos = 2520
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set vtx_channel = 5
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5820
set esc_sensor_halfduplex = ON

# profile
profile 0

# profile
profile 1

set d_pitch = 38
set d_roll = 34

# profile
profile 2

set p_pitch = 62
set d_pitch = 38
set p_roll = 56
set d_roll = 34

# restore original profile selection
profile 2

# rateprofile
rateprofile 0

set thr_mid = 20
set thr_expo = 35
set roll_rc_rate = 88
set pitch_rc_rate = 88
set yaw_rc_rate = 88
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 68

# rateprofile
rateprofile 1

set thr_mid = 20
set thr_expo = 35
set roll_rc_rate = 88
set pitch_rc_rate = 88
set yaw_rc_rate = 88
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 68
set rate_center_sensitivity = 80
set rate_center_weight = 25
set rate_end_weight = 25

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
```

**Setup / Versions**
 - Flight controller: Airbot F7
 - ESC: Airbot Furling32 4in1

nerdCopter, 2020-07-20:
* I don't really have a direct answer...
* Is this a new build that you are setting up, or one that worked correctly with betaflight?
* I know you have to discover sensors in opentx for sure.
* When i setup telem, i always have difficulty, but in general, nothing emuflight specific.

dentex, 2020-07-21:
Hello.
It's basically a new build because I changed the FC... The rest is the same. I don't know if the sensor discovery might be the issue here... I have to try, but thanks for the suggestion.
(Caveat: I have a problem on the radio, enter button not working. So I can't change settings on it. Let's close this for now)
Thanks.

closed 2020-07-21 by dentex