emuflight/EmuFlight #331: Resource remap for DJI osd

esquisseq, 2020-07-25:
So I’m using the betafpv aio 20a FC it only has serial_TX 1 pad but no RX 1 (TX/RX 2 are used for crossfire) I’m trying to remap motor 5 output to SERIAL_RX 2 by using the command:
RESOURCE MOTOR 5 NONE
RESOURCE SERIAL_RX 1 NONE
RESOURCE SERIAL_RX 1 B03

And nothing happens on EmuFlight, there’s no osd (I obviously enable msp on uart1). This exact command works perfect on betaflight. Is this a software bug, or remapping is not supported by EmuFlight?

nerdCopter, 2020-07-27:
resource mapping should work, but maybe it's more DJI incompatibility. EmuFlight has has no focus on it.
https://github.com/emuflight/EmuFlight/wiki/DJI

esquisseq, 2020-07-27:
That’s bad... Quad flies so much better on EmuFlight.

nerdCopter, 2020-07-27:
maybe ask in Discord... better chance of feedback
https://discord.gg/emfqENw