emuflight/EmuFlight #332: Missing EmuFlight target for HGLRC F722

davidj802, 2020-07-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

nerdCopter, 2020-07-31:
@davidj802 , are you in EmuFlight Discord? we have that target now, just not released.

davidj802, 2020-08-01:
@nerdCopter, Thanks just found target file on Discord.

closed 2020-08-14 by Quick-Flash