emuflight/EmuFlight #334: Resource remapping LED pin bricks Flywoo FC

fiveseven808, 2020-08-06:
**Describe the bug**
When I remap my resource pin from the default (incorrect) pin for FLYWOOF411 target, my FC gets bricked and needs to be reflashed

**To Reproduce**
Steps to reproduce the behavior:
Flash emuflight FLYWOOF411 0.3.1 on Flywoo 16x16 F4 stack. 
Map via:
resource pinio 2 none
resource led_strip 1 A15

**Expected behavior**
Since these assignments are from the betaflight firmware, the LED should work and nothing else should change. 

**Flight controller configuration**
The aforementioned change is the only diff

Kaiowarez, 2020-08-06:
So a15 is led strip and no cam control.. 🤔 I'll redo the timers
Led strip is assigned thru timers so it'll brick if you reassign it

nerdCopter, 2020-08-06:
@fiveseven808 , are you on Discord so we can send you.hex for testing?
https://discord.gg/gdP9CwE

nerdCopter, 2020-08-10:
* reported LEDs work.
* waiting for additional feedback make sure nothing else affected.

nerdCopter, 2020-08-11:
* reported failure
* see https://github.com/emuflight/EmuFlight/pull/335 for details

nerdCopter, 2020-10-18:
* fixed Aug 19 with https://github.com/emuflight/EmuFlight/pull/342

closed 2020-10-18 by nerdCopter