emuflight/EmuFlight #336: Missing Target for IFRC/IFLIGHT_F405_AIO(STM32F405)

brousselle, 2020-08-11:
As i do not like much where BF is going nowadays, i would like to join the emuflight community.
But i cannot flash emuflight on my AIO controller.

Can you please help, or tell me any workaround for this controller ?

Thanks in advance :D

nerdCopter, 2020-08-11:
`IFF4_AIO` is not the correct target?

brousselle, 2020-08-11:
Well this is what i see in the dump of BF:
# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F405_AIO, version: bc19b7dc, date: 2020-04-01T04:37:12Z

How can i check further or test this is the right target ?
Is there a risk if i try to flash it ?

nerdCopter, 2020-08-11:
if you have a boot-button, then there is no risk because you can always boot into DFU mode with the button. But i'm 99% sure `IFF_AIO` is what you want.  Emu does not necessarily use the same target names as betaflight.

brousselle, 2020-08-11:
Great news then !
I will try tomorrow and report back here. 
But i guess we can close the issue, i will reopen it if it does not work. 
Thanks for the quick reply !

closed 2020-08-11 by brousselle

brousselle, 2020-08-12:
confirmed: works like a charm :+1: