emuflight/EmuFlight #340: Some features not working properly in MAMBAF405US target

AndroidGX, 2020-08-14:
**Describe the bug**
Severals things not working into my FC (target: MAMBAF405US (S405)).
Not sure it's due to latest Emuflight Configurator, but for exemple the following things are not working as intended :

- MOTOR_STOP feature
- BIP when arming
- BIP when unarming

**To Reproduce**
Just activate those options on this target and confirm it's not working.
Activating or not as the same effect, it's like off by default.

**Expected behavior**
Those options should normally work.
I tried to compile myself from the target latest master branch from 3 days ago but same issue for this target :(
Let me know if I can flash a special build and confirm it will work again.

**Flight controller configuration**
`board_name 
manufacturer_id 
mcu_id 001e0040324d500220313256
signature 

# reset configuration to default settings
defaults nosave

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 A09
resource MOTOR 4 A08

# mixer

# servo

# servo mix

# feature
feature -DYNAMIC_FILTER
feature GPS
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 1 1 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 2 1350 2100 0 0
aux 2 46 2 1900 2100 0 0
aux 3 13 1 900 1100 0 0
aux 4 31 2 1400 1600 0 0
aux 5 35 1 1900 2100 0 0
aux 6 36 3 1900 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW180
set gyro_sync_denom = 2
set gyro_lowpass_hz_yaw = 115
set imuf_roll_q = 4000
set imuf_pitch_q = 4000
set imuf_yaw_q = 4000
set imuf_sharpness = 3000
set acc_calibration = 6,14,-49
set max_check = 2000
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 2
set failsafe_procedure = GPS-RESCUE
set align_board_roll = 180
set ibata_scale = 115
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 35
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1400
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 0
set deadband = 3
set yaw_deadband = 3
set osd_rssi_alarm = 45
set osd_cap_alarm = 0
set osd_rssi_pos = 2497
set osd_flymode_pos = 2106
set osd_gps_speed_pos = 2539
set osd_gps_lon_pos = 2048
set osd_gps_lat_pos = 2080
set osd_gps_sats_pos = 2465
set osd_home_dir_pos = 2272
set osd_home_dist_pos = 2273
set osd_altitude_pos = 2240
set osd_avg_cell_voltage_pos = 2516

# profile
profile 0

set dterm_lowpass_hz_roll = 100
set dterm_lowpass_hz_pitch = 100
set dterm_lowpass_hz_yaw = 100
set i_decay = 5
set p_pitch = 61
set i_pitch = 90
set d_pitch = 31
set p_roll = 55
set i_roll = 80
set p_yaw = 75
set i_yaw = 75
set d_yaw = 20

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 110
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 25
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 65

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; MAMBAF405US
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]; DJI FPV System (Air Unit)

nerdCopter, 2020-08-14:
i see this in your dump:
```
beeper -DISARMING
beeper -ARMING
```
which turns them off

i will ask devs about motor_stop, i never heard of anyone using motor_stop on a non-Whoop.

nerdCopter, 2020-08-14:
Also you do have a beeper soldered onto the FC correct?

AndroidGX, 2020-08-14:
> i see this in your dump:
> 
> ```
> beeper -DISARMING
> beeper -ARMING
> ```
> 
> which turns them off
> 
> i will ask devs about motor_stop, i never heard of anyone using motor_stop on a non-Whoop.
I took the dump after turning them back OFF  ;)
Yup, sometimes motor_stop can be useful also on 5inch (or more) quads depending on where they take off.

> Also you do have a beeper soldered onto the FC correct?
Nope, self-powered. I'm using ViFly Beacon (on my lipo strap with a tpu).

nerdCopter, 2020-08-14:
* okay, beeper command should only work wither soldered beeper/vifly (or clone) to FC.
* dhsot beeping for motors dont do all the beeper features.
* i use a vifly clone, my green wire is definitely on the beeper out of the FC, i forgot how the black/red were powered (but are also soldered on the FC) (but i also use a different FC, not Mamba).
* i'll link this ticket to devs/testers for feedback.

AndroidGX, 2020-08-14:
> * okay, beeper command should only work with soldered beeper/vifly (or clone) to FC.
> * dshot beeping for motors dont do all the beeper features.
> * i use a vifly clone, my green wire is definitely on the beeper out of the FC, i forgot how the black/red were powered (but are also soldered on the FC) (but i also use a different FC, not Mamba).
> * i'll link this ticket to devs/testers for feedback.

Ok thanks for clarifying, now I udnerstand better, I thought it was related to dshot features :)
So there is only 1 thing not working then, it's the MOTOR_STOP feature
Let me know if I can help you for any test

Quick-Flash, 2020-08-14:
I've never tested motor stop, so I dont know if that even works at all.

gretel, 2020-11-04:
stopping motors requires support `esc` side, too.

closed 2022-07-26 by nerdCopter