emuflight/EmuFlight #341: Adding a stock preset

AndroidGX, 2020-08-15:
Hey guys,

It could be really cool to have a stock preset in the PID section (like some other presets), so it will avoid each time reset the full data to remember what was the stock values in each cells.
(not sure if it's related to firmware or Configurator)

Many thanks and keep up the good work 👍

nerdCopter, 2020-08-17:
@AndroidGX , Emu has `Default-with-low-dTermLPFs` and `Default-with-high-dTermLPFs` which reset PIDs, SPA, TPA, boost, gyro LPF, dterm LPF, etc.

The only difference between the two are dTerm LPF's.  The "lower is 65,200 and the "higher" is 100,250.

AndroidGX, 2020-08-17:
Ah ok, never saw those profiles before, thank you very much :)

closed 2020-08-17 by AndroidGX