emuflight/EmuFlight #343: PIT mode via Modes does not seem to work

fiveseven808, 2020-08-21:
**Describe the bug**
When assigning PIT Mode to an AUX channel in the Modes tab, it does not switch to PIT mode when the aux switch is in the correct position. 

PIT mode appears to work properly via the OSD menus though.

**To Reproduce**
Steps to reproduce the behavior:
Set PIT mode to an aux channel. Try a ways away without a VRx antenna and toggle in and out of pit mode with the aux channel to confirm it working or not. 

**Expected behavior**
I expect the PIT mode engages when AUX channel is set, and disengage when not. 

**Flight controller configuration**
I've tried it on 3 flight controllers: FLYWOOF411, CRAZYBEEF4FR, OMNIBUSF4SD, all with the same results

**Setup / Versions**
VTx is built in for Crazybee F4 lite board, otherwise the NamelessRC Nano400 VTx was used to test this feature (IRC Tramp for all 3)

**Additional context**
VTx info OSD overlay does not seem to let me know when I'm in PIT mode, even when manually selected via OSD menu. This might just be a thing from old betaflight though. I never played too much with the pre 4.0 betaflights.

gretel, 2020-10-17:
isnt the powerlevel shown as `0` on pitmode?

fiveseven808, 2020-10-18:
No, it does not show 0 when pit mode is engaged.

nerdCopter, 2020-12-05:
does this commend help? `set vtx_low_power_disarm = ON`
i dont know if the is bewbeedrone specific or not, but using it on my whoop.

closed 2022-07-26 by nerdCopter