emuflight/EmuFlight #354: Enabling "DYNAMIC_FILTER" makes the quad going crazy once take off

AndroidGX, 2020-09-13:
**Describe the bug**
For an unknown reason, I'm now obliged to disable "DYNAMIC_FILTER".
If I turn it ON again, my 5 inch quad arming is OK at idle but once I take off it immediately going crazy and it's like hies going to try to flip itself, so I desarm immediately to avoid any damage to the propellers or other components.
I tried to be with stock config options (adapted to my FC) and same issue.
On my beginning on Emuflight I never had such issue with stock (unmodified) 0.3.1 firmware. Only had this issue since I'm compiling myself my own targets of unofficial 0.3.1 (latest master branch commits).

**To Reproduce**
Steps to reproduce the behavior:
Just turn ON "DYNAMIC_FILTER" from latest Emuflight configurator.

**Expected behavior**
Quad should take off normally instead of going crazy.

**Flight controller configuration**
https://pastebin.com/thXTZ2XB

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; MAMBAF405US

Many thanks in advance for your help,
Let me know if I can flash any specific firmware to help
Cheers'

nerdCopter, 2020-09-13:
It's absolutly a tuning issue. you will need to tune gyro and dterm LPF (lowpass filters) as per blockbox.  
If it want to rocket wither your LPF's are not in good spot or your P/D ratio is not a good tune either.

for what it's worth, generall with dynamic_filter enable (Matrix) ON, then you can use one dTerm LPF. If you have dynamic_filter OFF, then you generally need two dterm LPFs.

First step is to lowere your P's and D's significantly, record some Blackbox and find where you frequency noises are locatied and tune LPF's per such.  Then you can tune your PID afterward.

Join discord for further assistance.

AndroidGX, 2020-09-13:
Ok interesting, thank you so much for your help, will do that
Cheers'
👍

closed 2020-09-13 by AndroidGX

nerdCopter, 2020-09-13:
this assumes your modifications are not involved in the issue.  Welcome and good luck!!