emuflight/EmuFlight #359: Led strip not working after updating to Emuflight

SinnedFpv, 2020-09-24:
Hello there

I have an addressable LED strip installed on my cinewhoop. It was working fine when I had BF 4.2 but after I updated the firmware to EMU I have not been able to make it work again.

The led was a solid white during the installation of the new firmware.

I followed all the usual step to set up an LED strip:
-switch on the function from the conf tab
-wiring ordering mode
-Assign function to the numbered led
I also used CLI to enter the commands manually

Any idea on what the issue could be?
 
I have to say that I love the feeling of EMU on my cine and it has improved the problem I had with wobbles at low altitude but is just bugging me that I cannot use my led anymore

thanks everyone for the help!

Kaiowarez, 2020-09-24:
hey can you jump on discord?
and whats the fc target you used in emu ?

SinnedFpv, 2020-09-24:
Hey thank you for the reply!
I have flashed it with the version MAMBAF405US.
Sure I am on discord right now, what's the name of the conversation?

SinnedFpv, 2020-09-24:
I am going to sleep mate. Pretty late here sorry, If you like to have a
look i ll just paste my dump here.
Thank you for your help! 🙏

all

# version
# EmuFlight / MAMBAF405US (S405) 0.3.1 May 19 2020 / 04:42:42 (9366544) MSP
API: 1.47

board_name
manufacturer_id
mcu_id 002c00233152501220333647
signature

# name
name Sinned

# resources
resource BEEPER 1 C13
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
[... 1008 lines cut ...]
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set rate_center_sensitivity = 100
set rate_end_sensitivity = 100
set rate_center_correction = 10
set rate_end_correction = 10
set rate_center_weight = 0
set rate_end_weight = 0
set tpa_rate_p = 75
set tpa_rate_i = 125
set tpa_rate_d = 65
set tpa_breakpoint = 1600
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37
set vbat_comp_throttle_level = 75
set vbat_comp_pid_level = 75

nerdCopter, 2020-09-24:
@SinnedFpv https://discord.gg/gdP9CwE

Kaiowarez, 2020-09-24:
Ah I see your resource led_strip is showing none..
I fixed that issue for someone before, it's just not on current release if
you hop on discord we can make the target for you with the working led strip

nerdCopter, 2020-09-25:
Solved by re-flashing BetaFlight, then flashing latest EmuFlight build.

closed 2020-09-25 by nerdCopter