emuflight/EmuFlight #360: f7xsd flight controller (jhef7dual target)

gbbulciolu, 2020-09-26:
No gyro available, impossible to calibrate. On BF, same target (jhef7dual) works as expected.
Board: F7-XSD, same as this 
[image]

nerdCopter, 2020-09-27:
@Kaiowarez 
http://jhemcu.com/e_productshow/?19-JHEMCU-Betaflight-F7-XSD-2020mm-FC-OSD-5V-9V-BEC-19.html

nerdCopter, 2020-09-27:
@Kaiowarez , i'm pretty sure these are clones of other products, so maybe "original" targets are to be used?
`BETAFLIGHTXXX` ?

gbbulciolu, 2020-09-27:
Probably clones, but on BF with same target has no issues and works as expected.

Kaiowarez, 2020-09-30:
Problem is betaflight auto detects..
Try to change default gyro
Coz this fc only have 1 gyro

gbbulciolu, 2020-09-30:
problem is on emuflight, on bf it works like a charm.

Kaiowarez, 2020-09-30:
Get me a dump from bf and I'll see what is the issue

sebas1984x, 2020-10-12:
I can't get the osd voltage right, when plugged in a 3s it still says 4s. I calibrated but tha doesn't help.

nerdCopter, 2020-10-12:
@sebas1984x , there should be a better solution, but `set force_battery_cell_count = 4` might get you airborne.  I personally have no answer to why it's not working for you.

gbbulciolu, 2020-10-16:
> info from internal discussions: `gyro is mpu ... just needs to set use gyro 2`
> i.e. BeF unified targets finds mismatched hardware and fixes internally, our targets are static and this FC is a modified clone.

thanks for the tip... the correct cli command is: set gyro_to_use = second (possible choices: both, first, second).
cheers

majoe17, 2020-11-08:
> I can't get the osd voltage right, when plugged in a 3s it still says 4s. I calibrated but tha doesn't help.

Hello, have you solve this one?

nerdCopter, 2020-11-08:
did cli command help? `set force_battery_cell_count = X`  @sebas1984x ? 
@Kaiowarez might have to look into it further.  are you running 0.3.2 @majoe17 ?

sebas1984x, 2020-11-08:
Nope, can't get it right. Tried everything

Kaiowarez, 2020-11-08:
I never modified this target I looked at it tho..  It's a legacy target I could rewrite it maybe tomorrow the earliest

majoe17, 2020-11-08:
@nerdCopter i used all available firmware. Emu and betaflight. Still no luck. I already did that force count cell, still no luck.

majoe17, 2020-11-08:
I plugged in a 2s lipo but the configurator says 7s. Did some calibration, 110 down to 29, then whenever i plugged in a fresh, not fresh or almost empty 2s lipo, it will show the voltage i used to calibrate.

nerdCopter, 2020-11-09:
@majoe17 , @sebas1984x ... if even betaflight fails, then i would contact the vendor for fix, instruction, or even a replacement. maybe they have know hardware problems. 
Do you agree @Kaiowarez ?

closed 2020-12-05 by nerdCopter

nerdCopter, 2022-10-25:
> Hello, has any new information about fix wrong voltage? I try last EmuFlight and BetaFlight - broblem is the same.

no news. i can only recommend calibrate, but if it fails, then i have no other solution.

https://youtu.be/ZbyuPCcXh-8

[image]