emuflight/EmuFlight #363: VTX Table Smart Audio BetaFPV Beta85x 4K

siumam, 2020-09-27:
i've flash my Beta95x 4K firmware to latest emuflight firmware

everithing is perfect config but there is only one problem

VTX Table not correct and not work propertly
specially VTX Power table, i use beta diamond VTX as default there is only 200mw power.
but in Smart Audio VTX Power 0 1 2 3 4 is 25 200 400 600 800

how to change that table?
i try to use cli commande VTX Table smart audio but not work

> vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200
save

nerdCopter, 2020-09-28:
Emu 0.X.Y does not have vtx tables at all.  Emu was forked from Butterflight which was forked from betaflight 3.5.X
You may use smart-audio or physical VTX buttons.

siumam, 2020-09-28:
> Emu 0.X.Y does not have vtx tables at all. Emu was forked from Butterflight which was forked from betaflight 3.5.X
> You may use smart-audio or physical VTX buttons.

there is no Physical VTX button in my BetaFPV Diamond VTX
in Smart Audio OSD i can change power with default power table, VTX Power Table was not correct
when i choose 200mw that lvl. 2 power i feel like 25mw power.
should i choos maximum lvl. power as 800mw?

nerdCopter, 2020-09-29:
you may experiment with each. i dont have immediate solution.
 Emu is fork of old betaflight 3.5.x. new rebase will be in far future, not soon.

closed 2022-04-01 by BeauBrewski