emuflight/EmuFlight #368: Change vtx power level using aux channel

stkenneth, 2020-09-30:
please add a feature in the adjustment aux mode to change vtx power level or channel or band through aux channel. betaflight has finally implemented this on betaflight 4.x
"
For example the following will configure a 3-position switch on Aux3 to switch power levels 1/2/3.

vtx 0 2 0 0 1 900 1200 vtx 1 2 0 0 2 1300 1700 vtx 2 2 0 0 3 1800 2100 

This can also be used to switch to specific band/channels based on a switch but since the band/channel in the example above are 0 it won't change channels - only power. See the help for the vtx command.

"
reference : 
https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

Igorshp, 2020-12-04:
Good news!  PR is on it's way :)

https://github.com/emuflight/EmuFlight/pull/417

nerdCopter, 2020-12-05:
@StonedDawg , would you like a .hex to test this? just let us know the target.
This is not our normal testing route, but i think we have few testers that use this.

StonedDawg, 2020-12-06:
> @StonedDawg , would you like a .hex to test this? just let us know the target.
> This is not our normal testing route, but i think we have few testers that use this.

ah ok i think i can do that, target is fury4osd

nerdCopter, 2020-12-06:
@StonedDawg [FURYF4OSD_test_vtx_power_aux.zip](https://github.com/emuflight/EmuFlight/files/5649346/FURYF4OSD_test_vtx_power_aux.zip)

closed 2021-01-04 by nerdCopter

StonedDawg, 2021-01-15:
@nerdCopter thankyou for looking into this.... sorry I havent get back soon enough.... i keep trying to switch pad on fc and fail but it turned out that my vtx UART port is dead.. I will get back to this once I have new vtx..

nerdCopter, 2021-01-15:
@StonedDawg , we merged the code. it's in 0.3.3 release which is live and available to everyone. thank you.