emuflight/EmuFlight #405: [RFE] Allow change failsafe behavior with an AUX/stick

soukron, 2020-11-12:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I must connect a laptop or a phone to my FC to change the behavior of the failsafe to either enable RTH or drop when I could use a stick to change it.

**Describe the solution you'd like**
I would be awesome if we could change with one stick the behavior of the failsafe stage 2 to return to home, drop or land. That would allow us to change easily when using the same drone for different flights without connecting a latop to it.

**Describe alternatives you've considered**
There's no alternative. The only way to change this setting is to connect the drone to a laptop or phone and change it.

**Additional context**
As said, allowing changes with sticks would be perfect for this setting.

This could be done using modes or assigning a channel (an AUX) to the stage 2 failsafe procedure in the failsafe settings page and then setting a range to each procedure (drop, land or gps rescue).

Note that this request is not only for digital users (which don't have access to the OSD) but also situations where you forgot to change the setting before taking off or if you are already flying and change your mind about flying far away (so you need RTH) or entering in a bando that you find and change the setting to avoid a RTH in a closed room.

Quick-Flash, 2021-04-02:
This could be a potential safety concern if you change modes unintentionally.

soukron, 2021-04-05:
One could say the same about disarming unintentionally... but no worries, I already had weeks of conversations with the betaflight team and they rejected the RFE so I'd expect the same from here.

closed 2021-04-05 by soukron