emuflight/EmuFlight #411: Tbs crossfire vbat telemetry issue

0001gravity, 2020-11-18:
I'm using an Omnibus f4 v6 fc and I'm using a Taranis x9d with a crossfire unit.  On the telemetry page, I only have
1- 1rss
2- 2rss
3-rqly
4-rsnr
5-rfmd
6-tpwr
7-trss
8-tqly
9-tsnr

I'm trying to get the vbat value so I can set up a low voltage alarm. The fc is using the .most recent firmware, I've also deleted and rediscovered sensors.
Everything else check out but I can't figure put how to get the voltage value.
On betaflight and on the OSD the voltage is shown.

nerdCopter, 2020-11-19:
@0001gravity , i've asked some insider-pilots, ...
only response i have so far, is that one pilot has no problem for vbat on `Omnibus f4 v6 fc` but he does not have CRSF.

https://github.com/emuflight/EmuFlight/wiki/CRSF-LQ

i'll reply if new info is learned.

nerdCopter, 2020-11-27:
any news @0001gravity ??
`VFAS` still not seen with s.port soldered, telem-enabled on config-tab, and UART on port-tab setup?

nerdCopter, 2020-12-07:
i've had further feedback:
```
[8:30 PM] Josh G: it's working fine for me after re-setting up my sensors. Honestly sounds like he has telemetry disabled in the configuration tab. As long as he is on crossfire 4.08 it should work fine.
[10:25 PM] Josh G: The correct sensor is RxBt
```

closed 2021-10-08 by nerdCopter