emuflight/EmuFlight #412: OSD not showing all options

thommicom, 2020-11-19:
Not all options are visible in the Emuflight configurator. E.g. I miss any kind of GPS item and other items.

To get around the problem, I can select the options I want to see via the transmitted image of the VTX. These then appear in the middle of the screen. Now I can use the Emuflight configurator to move the item to where I want it. The item is still not displayed in the selection list in the Configurator, but is still shown in the preview.

I have the problem on different flight controllers. The problem also occurs on a second computer. Reinstallation of EMuflight on the flight controller and reinstallation of the configurator were unfortunately unsuccessful.

[image]

Zlopi, 2020-11-19:
[image]

Try this enable its? you should very like it... =)))

Zlopi, 2020-11-19:
[image]

Maybe turn on GPS first?

thommicom, 2020-11-19:
F**k. Thats it. In Expert Mode the missing Items are visible. GPS was already turned on before. 

Thanks a million!

closed 2020-11-19 by thommicom