emuflight/EmuFlight #414: Matekf411SE Uart1 not working

dronotron, 2020-11-23:
On the matekf411SE target, Uart1 is unable to communicate with devices connected to it on emuflight 0.3.2 nor 0.3.1 (target was unavailable on previous versions)

I had a gps connected to that port and it never worked on emuflight, I suspected an incompatibility between the device and emuflight so I kept using betaflight on this fc, with the device working properly.

But yesterday I connected a dji airunit to that same port (Uart1), fc flashed with betaflight and it worked properly (osd working). If the fc is flashed with emuflight 0.3.2, the osd on the airunit  is gone. The only conclusion is this target has an issue with the implementation of this particular serial port.

Kaiowarez, 2020-11-23:
> Assigned #414 <https://github.com/emuflight/EmuFlight/issues/414> to
> @Kaiowarez <https://github.com/Kaiowarez>.

I'll fix it when I get home

nerdCopter, 2021-01-25:
closed by https://github.com/emuflight/EmuFlight/pull/474

closed 2021-01-25 by nerdCopter