emuflight/EmuFlight #415: baro BMP280 not detect FURYF4OSD

maxgame91, 2020-11-24:
Hello,

I tried to use an external barometer, a BMP280 on my mamba f405 mk2 so I ha ve for target the FURYF4OSD, but my baro is not detect by the board.

I tried on betaflight 4.2 (need a custom unified target). So wiring is not the problem here.

In the CLI i have nothing when I type "get baro" I have nothing related to a barometer.

Do i need to modified something in the target or somewhere else ? My goal is to use [this branch](https://github.com/emuflight/EmuFlight/tree/add_alti_limit) to have an alti limiter

nerdCopter, 2020-11-24:
@Kaiowarez ?

Kaiowarez, 2020-11-24:
Hmm might need to add baro drivers

maxgame91, 2020-11-24:
@Kaiowarez What file I need to modify ? And what I need to add ?

Thanks

maxgame91, 2020-11-25:
Hi guys, 

So when I add this line "drivers/barometer/barometer_bmp280.c" in the file "[target.mk](https://github.com/emuflight/EmuFlight/blob/master/src/main/target/FURYF4/target.mk)" in the FURY4 target folder in the line 18,
I got an error when I tried a "make FURYF4OSD" in bash https://imgur.com/a/aDaH9z9

I flash my board with the FURYF4 target but I don't have any osd so it's not really good

Do you have any lead ?

maxgame91, 2020-11-25:
Hi guys,

So when I add this line "drivers/barometer/barometer_bmp280.c" [like this](https://imgur.com/eoK6U94) in the [FURY4](https://github.com/emuflight/EmuFlight/blob/master/src/main/target/FURYF4/target.mk) target folder,
I got an error when I tried a "make FURYF4OSD" in bash https://imgur.com/a/aDaH9z9

I flash my board with the FURYF4 target but I don't have any osd so it's not really good

Do you have any lead ?

nerdCopter, 2020-11-26:
you'll need a `\` at the end of the prior line.

maxgame91, 2020-11-26:
@nerdCopter thanks I can now make my hex file for this target but still nothing on the configurator, I also modify the [target.h](https://imgur.com/TiDQgZ2) at line 92 and 93 but don't work.

nerdCopter, 2020-11-26:
@maxgame91 , So you're flashing the local .hex you made and your external Baro 280 is still not seen, correct?
i think the target.h mod implies an internal resource i2cdev_1, but i dont personally know. i would think you'd jsut want to add the driver, then assign the resource to uart in CLI.  but i've never done such a thing.
@Kaiowarez may have some ideas.

maxgame91, 2020-11-26:
@nerdCopter You are correct, I have [this branch](https://github.com/emuflight/EmuFlight/tree/add_alti_limit) on my computer and make my own .hex and yes my external bmp280 is not seen, enven when I add the driver in the ["target.mk"](https://github.com/emuflight/EmuFlight/blob/add_alti_limit/src/main/target/FURYF4/target.mk).

It look like the i2c are define in the right pin but I'm clearly not a dev so maybe I'm wrong

I'm really open to idea yes

Kaiowarez, 2020-11-26:
First you add the drivers on the. Mk file needed \ on it
Then on target.h
Need to confirm i2c is working
Baro i2c device
Baro spi device
Cs pin if needing assignment

Kaiowarez, 2020-11-26:
If it's on a uart make sure it's on either half or full duplex mode..
Need it to go in and out for baro to work

nerdCopter, 2020-11-26:
i only see ESC and GPS as sensor inputs, so maybe we are just not updated enough,  :man_shrugging: 
maybe a gps with built in baro is better solution?? i have no clue, so sorry. i never used either.
Kaio is the target guy, but i dont know if this is a target specific issue or not, i think it's more a lack of capability.
EmuFlight was forked from butterflight which was forked from betaflight 3.5.x.  if 3.5.x was capable, we should be also.
otherwise, maybe this jsut wont be possible until Emu 1.0.0  ?? :man_shrugging:

maxgame91, 2020-11-26:
I add the \
i2c is working I tried on betaflight 4.2 and my bmp280 was detect
For my board scl and sda are define by this pin : 

	#define I2C1_SCL                PB6        // SCL pad
	#define I2C1_SDA                PB7  

and you lost me at spi and cs pin sorry

Kaiowarez, 2020-12-03:
Yeah that's right
Then i2c_device needs to match which i2c it is
Then defined for baro to use then if baro has a cs pin its using and spi
instance as well needs to be defined

BeauBrewski, 2021-07-27:
Has any official/test updates with this driver been made? Got another pilot on the FB page with same problem:

> Hi all
> I need your help, I have FC mambaF405 and I'm using firmware FURYF4OSD 0.3.4, I installed an external BMP280 sensor but it doesn't detect the barometer, I've already tested it with the latest version of Betaflight and it detects the sensor as a barometer but not in Emuflight, therefore the connections are well made.
> Is there any way to enable the BMP280 sensor in Emuflight?

BeauBrewski, 2021-08-12:
And another
> hi every one. I have a problem with Flight controller Rush F7 Digital and Emuflight. My FC supports barometer BMP280 but I cant turn it on in Emu. Some one help me please.

nerdCopter, 2021-08-12:
i wonder if @JulioCesarMatias would be interested in this issue?

JulioCesarMatias, 2021-08-13:
> i wonder if @JulioCesarMatias would be interested in this issue?

yes, when i have time i will dedicate myself to it

JulioCesarMatias, 2021-08-13:
https://github.com/emuflight/EmuFlight/pull/671

is there anyone to test this?

nerdCopter, 2021-08-13:
[BMP280-TEST__EmuFlight_0.3.7_FURYF4OSD_Build_2010_eb578ff23.hex.zip](https://github.com/emuflight/EmuFlight/files/6984816/BMP280-TEST__EmuFlight_0.3.7_FURYF4OSD_Build_2010_eb578ff23.hex.zip)

BeauBrewski, 2021-08-13:
Do I need to make a separate bug fix request for the `RUSHBLADEF7` target then?

BeauBrewski, 2021-08-13:
> #671
> 
> is there anyone to test this?

Hopefully will hear results from one of the guys on FB page.

BeauBrewski, 2021-08-13:
Got a response. Baramoter is still not showing up on FURYF4OSD. See attached images (betaflight image included to show working):
[image]
[image]

BeauBrewski, 2021-08-14:
[BMP280 Test_EmuFlight_0.3.7_RUSHBLADEF7_Build_2013_3da929cc6.zip](https://github.com/emuflight/EmuFlight/files/6986691/BMP280.Test_EmuFlight_0.3.7_RUSHBLADEF7_Build_2013_3da929cc6.zip)

BeauBrewski, 2021-08-14:
[BMP230 Test2_EmuFlight_0.3.7_FURYF7_Build_2013_3da929cc6.zip](https://github.com/emuflight/EmuFlight/files/6986692/BMP230.Test2_EmuFlight_0.3.7_FURYF7_Build_2013_3da929cc6.zip)

BeauBrewski, 2021-08-14:
[BMP280 Test 2_EmuFlight_0.3.7_FURYF4OSD_Build_2013_3da929cc6.zip](https://github.com/emuflight/EmuFlight/files/6986709/BMP280.Test.2_EmuFlight_0.3.7_FURYF4OSD_Build_2013_3da929cc6.zip)

nerdCopter, 2021-08-14:
i cherry-picked code into 0.3.4: please test: 
[TESTBMP280__EmuFlight_0.3.4_FURYF4OSD_Build_3ec879187.hex.zip](https://github.com/emuflight/EmuFlight/files/6986824/TESTBMP280__EmuFlight_0.3.4_FURYF4OSD_Build_3ec879187.hex.zip)

BeauBrewski, 2021-08-15:
That last update got it working! 
[image]

nerdCopter, 2021-08-15:
@BeauBrewski , let me know if it works (has fair values) in the air as well, then i can merge to master.

BeauBrewski, 2021-08-15:
>  I confirm that it is working and give the correct values.
once again thank you very much for the support. 😉✌️👍

JulioCesarMatias, 2021-08-17:
problem solved, close this.

closed 2021-08-17 by nerdCopter