emuflight/EmuFlight #423: [Target Request] Rush Blade F7

nerdCopter, 2020-12-05:
Rush Blade F7

```
dump
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.4 Oct 20 2020 / 08:20:06 (fbcaf8c50) MSP API: 1.43
###ERROR: dump: NO CONFIG FOUND###
# start the command batch
batch start

board_name BLADE_F7
manufacturer_id RUSH

# name: -

# resources
resource BEEPER 1 B02
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 B01
resource MOTOR 4 B00
resource MOTOR 5 C07
resource MOTOR 6 C06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
[... 1019 lines cut ...]
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
```

nerdCopter, 2021-01-28:
closed by https://github.com/emuflight/EmuFlight/pull/474

closed 2021-01-28 by nerdCopter

MiniUAV11, 2021-12-06:
Hello I have a Rush blade F722 flight controller the analog version, would this cli dump be the solutions for the unified target STM32F7X2 not working? If so would I just copy and paste this in the CLI tab in betaflight or no? I am not as experienced with computer stuff.

nerdCopter, 2021-12-06:
flash `RUSHBLADEF7`
[image]
but save a `dump all` from CLI from BetaFlight first in case you have to revert.

MiniUAV11, 2021-12-06:
Would I flash the information on betaflight by loading this information onto the firmware flasher?

nerdCopter, 2021-12-06:
> Would I flash the information on betaflight by loading this information onto the firmware flasher?
no, do not paste the betaflight text into emuflight.

MiniUAV11, 2021-12-06:
I use betaflight I do not use emuflight

nerdCopter, 2021-12-06:
> I use betaflight I do not use emuflight
okay.
this dump was originally taken from betaflight, so your flash+defaults would be the same as this.
i dont use betaflight. this is an emuflight oriented github.

MiniUAV11, 2021-12-06:
So this cli is the same as what comes preloaded on the board when you buy it? Because that is what I am looking for since I cannot find the correct target and loaded the unified target from the manual onto my board and bricked it.

nerdCopter, 2021-12-06:
> So this cli is the same as what comes preloaded on the board when you buy it? Because that is what I am looking for since I cannot find the correct target and loaded the unified target from the manual onto my board and bricked it.

this would be the resources/settings for defaults.  i dont know what version of BF your's came with, but this dump is `4.2.4 `.

you mostly can't brick an FC, just press and hold the boot button while plugging in USB, then the Configurator will see `DFU` mode and you can flash again.

MiniUAV11, 2021-12-06:
I'm just looking to make my board work until betaflight supports this board. But I then just copy and paste this, load this into the flasher, and flash my board with this right?

nerdCopter, 2021-12-06:
> I'm just looking to make my board work until betaflight supports this board. But I then just copy and paste this, load this into the flasher, and flash my board with this right?

f you already have a betaflight fashed and you can get to CLI, then sure, try this out. you can't truly brick it.  you can always perform the boot-button trick to reflash.

essentially you can only flash .hex or .bin files.  in your case, you flash `STM32F7X2`and betaflight configurator should take care of the rest.

this text dump is simply list of configurations for that firmware.  maybe best to ask betaflight support.  they have a "Slack" https://slack.betaflight.com/