emuflight/EmuFlight #428: MAMBAF722_I2C

spar444, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

nerdCopter, 2020-12-07:
please send cli `dump` from betaflight or link to unified target.
please send link to official product specs.

spar444, 2020-12-07:
Targer - https://cdn.shopify.com/s/files/1/0027/2708/4144/files/MAMBA_F722_I2C_0716.config?v=1603377151
Dump - https://drive.google.com/file/d/1kT04KVcL_HGbSeHp_CqFxXto2mldSXOk/view?usp=sharing
Link to official store - https://www.diatone.us/collections/flightcontroller/products/mamba-f722s-flight-controller

nerdCopter, 2020-12-07:
@spar444 would you like to join Discord and @nerdCopter or "@Kaio Pryde" to test the target?

spar444, 2020-12-07:
I'm already there:)

nerdCopter, 2021-01-28:
closed by https://github.com/emuflight/EmuFlight/pull/474

closed 2021-01-28 by nerdCopter