emuflight/EmuFlight #429: [Target Request] MATEKF405SE

nerdCopter, 2020-12-07:
http://www.mateksys.com/?portfolio=f405-se

```
# version
# Betaflight / STM32F405 (S405) 4.2.5 Nov 22 2020 / 18:38:23 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405SE, version: 42dd44a1, date: 2020-07-24T06:02:46Z

# start the command batch
batch start

board_name MATEKF405SE
manufacturer_id MTKS

# name: kosmik FPV

# resources
resource BEEPER 1 C15
resource MOTOR 1 B07
resource MOTOR 2 B06
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 C08
resource MOTOR 6 C09
resource MOTOR 7 B14
resource MOTOR 8 B15
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A02
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
[... 1044 lines cut ...]
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 70
set yaw_rc_rate = 135
set roll_expo = 30
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 76
set pitch_srate = 83
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
```

nerdCopter, 2021-01-28:
* closed by https://github.com/emuflight/EmuFlight/pull/474

BeauBrewski, 2021-06-28:
Can close, has a official release 0.3.4 target.