emuflight/EmuFlight #430: [Target Request]  Lumenier LUX F7 HD "LUXF7HD"

nerdCopter, 2020-12-08:
* Lumenier LUX F7 HD "LUXF7HD".
* requested by @lemirec
* https://github.com/betaflight/unified-targets/blob/master/configs/default/LMNR-LUXF7HDV.config
* https://www.getfpv.com/lumenier-lux-hd-lite-f7-flight-controller-dji-hd-system-ready.html

lemirec, 2020-12-08:
Thanks

  *   https://github.com/betaflight/unified-targets/blob/master/configs/default/LMNR-LUXF7HDV.config<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fbetaflight%2Funified-targets%2Fblob%2Fmaster%2Fconfigs%2Fdefault%2FLMNR-LUXF7HDV.config&data=04%7C01%7C%7C72f32378a3c24a7abe1108d89b9a1c73%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C1%7C637430433493441041%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=kJK%2BnAdSQKxpTubq4Uj%2FI2ERJGi2z%2BrOJi6SmxCxSVk%3D&reserved=0>
  *   https://www.getfpv.com/lumenier-lux-hd-lite-f7-flight-controller-dji-hd-system-ready.html<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.getfpv.com%2Flumenier-lux-hd-lite-f7-flight-controller-dji-hd-system-ready.html&data=04%7C01%7C%7C72f32378a3c24a7abe1108d89b9a1c73%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C1%7C637430433493441041%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=34tDITAJ9dgpMuT3nhRFO1hLGIzn4ZMlDtzhrNzK%2BBs%3D&reserved=0>

BeauBrewski, 2021-08-24:
@Kaiowarez test file needed.

BeauBrewski, 2021-11-12:
Needs tested!
Check gyro orientation, motor arrangement, flash, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY
[EmuFlight_0.4.0_LUXF7HDV_Build_1f493948a.zip](https://github.com/emuflight/EmuFlight/files/7529145/EmuFlight_0.4.0_LUXF7HDV_Build_1f493948a.zip)

lemirec, 2021-11-16:
Nice, I’ll fly it first thing tomorrow morning.
Thanks, for your help!
Thank you
Chris

BeauBrewski, 2022-03-10:
@lemirec 
Any updates of functionality of this target?
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer 
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features, issues you noticed?

jbenedetto, 2022-07-26:
can't flash through betaflight.. restarts in boot mode, then does nothing.

Kaiowarez, 2022-07-26:
Need to flash with emuflight configurator

PR #1032 (2024-04-18): [target] add LUXF7HDV
resolves #430
closed 2024-04-18 by nerdCopter