emuflight/EmuFlight #433: [Target Request] Eachine Cvatar CineWhoop

nerdCopter, 2020-12-11:
**Vendor/Model**
`Eachine Cvatar CineWhoop`
`Eachine F722 mini Flight Controller`
https://www.eachine.com/Eachine-Cvatar-142mm-6S-3-Inch-Cinewhoop-Duct-FPV-Racing-Drone-PNP-or-BNF-Analog-Version-RunCam-Nano2-Cam-F722-DJI-FC-1507-2400KV-35A-ESC-Without-receiver-p-2013.html

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/tree/master/configs/default -- maybe EACH-EACHINEF722.config or EACH-EACHINEF722_AIO.config , need to verify which or none

```
# 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.5 Nov 22 2020 / 18:39:53 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: EACH, board_name: EACHINEF722, version: f1df40a9, date: 2020-07-14T08:09:18Z

# start the command batch
batch start

board_name EACHINEF722
manufacturer_id EACH

# name: Eachine Cvatar

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
[... 1049 lines cut ...]
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 90
set pitch_rc_rate = 85
set yaw_rc_rate = 80
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 45
set roll_srate = 70
set pitch_srate = 68
set yaw_srate = 64
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
```

bigjoncoop, 2021-03-28:
also been waiting for Emuflight target for the Eachine Cvatar EACHINEF722 .  

ALSO KNOW AS "STM32F7X2" BUT AFTER FLASHING EMU TARGET "STM32F7X2"  TO THIS FLIGHT CONTROLLER, USB NO LONGER WORKS. NOT DETECTED BY PC. MUST FORCE DFU MODE

[image]

nerdCopter, 2021-03-28:
dont flash STM32FX7, that one is not a typical generic target, it's for testbeds.
i think the eachine 722 has been requested, but not yet defined.

BeauBrewski, 2021-08-24:
@Kaiowarez test file needed.

BeauBrewski, 2021-11-13:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_EACHINEF722_Build_861a8d322.zip](https://github.com/emuflight/EmuFlight/files/7531903/EmuFlight_0.4.0_EACHINEF722_Build_861a8d322.zip)

PR #1033 (2024-04-18): [target] add EACHINEF722
resolves #433
closed 2024-04-18 by nerdCopter