emuflight/EmuFlight #434: [Target Request] GEPRCF722

nerdCopter, 2020-12-11:
Requested by "@Josef Dahari" discord

**Vendor/Model**
`GEPRCF722`
waiting on requester for which:
https://geprc.com/product/gep-f722-dual-bt/
or
https://geprc.com/product/gep-f722-hd-flight-controller/

**CLI dump or Unified Target**
 here: https://github.com/betaflight/unified-targets/tree/master/configs/default

BeauBrewski, 2021-08-24:
@nerdCopter @Kaiowarez 
Can we get a 0.4.0 update for this test file?
https://cdn.discordapp.com/attachments/691022009865338950/875181723787337788/EmuFlight_0.3.7_GEPR_GEPRCF722_Build_uncommitted_20210811EDT.hex

nerdCopter, 2021-08-25:
if @Kaiowarez  has the code pushed to his github, then i could retrieve it.

BeauBrewski, 2021-11-13:
Needs tested!
Check gyro orientation, motor arrangement, flash, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_GEPRCF722_Build_0ebfae7d7.zip](https://github.com/emuflight/EmuFlight/files/7532143/EmuFlight_0.4.0_GEPRCF722_Build_0ebfae7d7.zip)

PR #1034 (2024-04-18): [target] add GEPRCF722
resolves #434
closed 2024-04-18 by nerdCopter