emuflight/EmuFlight #449: CrazyBeeF4FS Bind command not found

TheMatrix97, 2020-12-25:
Hi!
Today I flashed the latest target for my CrazyBeeF4FS. When I tried to run the command "bind" via CLI I'm getting a solid "Command not found"
Thankfully is not a problem for me, because when I flashed my quad automatically entered in bind mode. But I don't want to reflash my quad in case I wanted to change of TX.....

**To Reproduce**
Flash latest emuflight release
Run "bind" in CLI tab

**Expected behavior**
getting a "binding rx..." output

**Flight controller configuration**
It's a fresh new install, nothing changed

**Setup / Versions**
- CrazyBeeF4
- UZ65 tiny

[image]

nerdCopter, 2021-01-04:
i dont understand why some FC have `bind` proper and other dont.
a work around is copy/paste cli items from betaflight and it should connect to pre-existing opentc profile.

for example my FrSky spi has the following iems that i used from a BF dump:
```
set frsky_spi_tx_id = 
set frsky_spi_offset = 
set frsky_spi_bind_hop_data = 
```

TheMatrix97, 2021-01-04:
@nerdCopter Thanks for the answer, I see.... Is emuflight using the same targets from bf? I guess there may be some info wrong or empty for this specific target....
Is there any way to set the fc in debug mode so I can proporcionate more info?

nerdCopter, 2021-01-04:
@TheMatrix97 , each target was either re-created or inherited from old code.
EmuFlight was forked from Butterflight which was forked from Betaflight 3.5.x so the whole code-base is stale, but Emu is working on a new re-base for the future.  However, such will not be soon.

I dont personally know of any CLI debug modes.

your two best bets are physical bind button or copy/paste previous BF bind data

TheMatrix97, 2021-01-04:
@nerdCopter ok, Thank you very much. I'm closing the issue

closed 2021-01-04 by TheMatrix97