emuflight/EmuFlight #456: MATEKF411MINISE

jazzyboy, 2021-01-01:
Please provide the target name. e.g. `MATEKF411MINISE`
Please provide vendor and model of the FC. e.g. `MATEKSYS  F411-MINI SE Flight Controller`
Please provide a lint to the vendor specifications.
http://www.mateksys.com/?portfolio=f411-minise

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF411SE.config

The BF 4.x flash is ok on my iflight Megabee quad with the target MTKS-MATEKF411SE, but the target doesn't ok with emuflight 0.3.2.
Nota - I'm not yet used to modifying / creating targets myself

jazzyboy, 2021-01-28:
Have you been able to advance on this target ?
I have tried this evening with the last build 0.3.3 but i could see there is no change on the behavior.
Thanks

nerdCopter, 2021-01-28:
@jazzyboy ,
Mr. @Kaiowarez does these in his spare time.  We all volunteer our personal time.  no ETR.  
He is working on weekends toward "all targets" though.
If you know anyone who could contribute, we welcome such.

jazzyboy, 2021-01-28:
Thanks i would be interested by helping the Emuflight but i don't see really well how to analyse and solve a target issue. I've just began to work on the Emuflight code with a fork on my github. I'm a french pilot and i would work on a new altitude limiter for the french regulation (as loutwice made with his project). Since I worked first on the integration of the betaflight feature named POSITION_ALT_SOURCE, to force the use of the barometer when there is a gps too on the build. After i think I'll take example on the altitude management in the RTH for altitude limiter. Unfortunately I don't have a lot of time to develop code.

nerdCopter, 2021-06-03:
closing this. `MATEKF411SE` was added Jan 25.
However, in hindsight, there may be issue with ADC pin/channel.  please let me know if this your experience.

closed 2021-06-03 by nerdCopter