emuflight/EmuFlight #459: NewBeeDrone Hummingbird Pro with EMU v0.3.3 - D16 (FRSKY_X) rxloss

jgauruder, 2021-01-03:
**Describe the bug**
rxloss at relatively short indoor distance when using FRSKY_X (D16)

**To Reproduce**
load emu 0.3.3 on NBD Hummingbird Pro - i tried EmuFlight_0.3.3_NBDHMBF4PRO.hex and EmuFlight_0.3.3_HMBF4PRO_US.hex
 configure SPI receiver with FRSKY_X selected

**Expected behavior**
no rxloss is expected within indoor single family home.

**Flight controller configuration**
'''
# 

# diff all

# version
# EmuFlight / NBDHMBF4PRO (HMBF4 PRO) 0.3.3 Dec 29 2020 / 00:15:31 (a456436e4) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 002600223030510435393433
signature 

# reset configuration to default settings
defaults nosave

# name
name EMU_3.3 HMB_Pro

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -LED_STRIP
feature -AIRMODE
feature -DYNAMIC_FILTER

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1900 2100 0 0
aux 1 1 2 925 1100 0 0
aux 2 2 2 1400 1600 0 0
aux 3 13 1 1400 1600 0 0
aux 4 28 2 1400 2100 0 0
aux 5 30 2 925 1600 0 0
aux 6 35 3 1900 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxrange
rxrange 0 988 2012
rxrange 1 988 2012
rxrange 2 988 2012
rxrange 3 988 2012

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz_yaw = 90
set imuf_roll_q = 4000
set imuf_pitch_q = 4000
set imuf_yaw_q = 4000
set imuf_w = 16
set imuf_sharpness = 6000
set acc_calibration = 45,-87,-175
set min_check = 1005
set max_check = 1995
set fpv_mix_degrees = 90
set cinematic_yaw = ON
set airmode_start_throttle_percent = 40
set blackbox_device = NONE
set min_throttle = 1035
set rx_spi_protocol = FRSKY_X
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 33
set yaw_motors_reversed = ON
set crashflip_power_percent = 100
set runaway_takeoff_prevention = OFF
set osd_vbat_pos = 2360
set osd_rssi_pos = 2208
set osd_tim_1_pos = 2072
set osd_tim_2_pos = 2336
set osd_flymode_pos = 2426
set osd_throttle_pos = 2368
set osd_vtx_channel_pos = 2304
set osd_current_pos = 2391
set osd_craft_name_pos = 2408
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2400
set vtx_channel = 1
set vtx_freq = 5658
set frsky_spi_tx_id = 148,219
set frsky_spi_offset = 93
set frsky_spi_bind_hop_data = 39,62,37,52,17,75,48,13,29,44,28,41,12,46,74,19,5,57,72,27,38,6,60,32,65,26,16,8,50,23,56,35,40,14,7,61,53,15,67,4,25,9,58,73,63,55,64,0,0,0
set frsky_x_rx_num = 1

# profile
profile 0

set smart_dterm_smoothing_roll = 50
set smart_dterm_smoothing_pitch = 50
set witchcraft_roll = 0
set witchcraft_pitch = 0
set spa_roll_i = 75
set spa_pitch_i = 75
set spa_yaw_i = 45
set crash_recovery = ON
set iterm_rotation = OFF
set throttle_boost = 0
set feathered_pids = 0
set i_decay = 7
set emu_boost = 0
set emu_boost_yaw = 0
set emu_boost_limit = 0
set emu_boost_limit_yaw = 0
set p_pitch = 95
set i_pitch = 85
set d_pitch = 85
set p_roll = 95
set i_roll = 85
set d_roll = 85
set p_yaw = 120
set i_yaw = 85
set d_yaw = 0
set d_angle_low = 35
set p_angle_high = 80
set level_limit = 80
set horizon_tilt_effect = 0

# profile
profile 1

set smart_dterm_smoothing_roll = 50
set smart_dterm_smoothing_pitch = 50
set witchcraft_roll = 0
set witchcraft_pitch = 0
set pid_at_min_throttle = OFF
set spa_roll_i = 75
set spa_pitch_i = 75
set spa_yaw_i = 45
set crash_recovery = ON
set iterm_rotation = OFF
set throttle_boost = 0
set feathered_pids = 0
set i_decay = 7
set emu_boost = 400
set emu_boost_yaw = 0
set emu_boost_limit = 55
set emu_boost_limit_yaw = 0
set p_pitch = 85
set i_pitch = 75
set d_pitch = 75
set p_roll = 85
set i_roll = 75
set d_roll = 75
set p_yaw = 100
set i_yaw = 75
set d_yaw = 0
set p_angle_low = 25
set d_angle_low = 35
set p_angle_high = 10
set level_limit = 80
set horizon_transition = 80
set horizon_tilt_effect = 0

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 205
set roll_expo = 24
set pitch_expo = 24
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 22
set rate_center_sensitivity = 55
set rate_end_sensitivity = 125
set rate_center_correction = 25
set rate_end_correction = 75
set rate_center_weight = 50
set rate_end_weight = 40

# rateprofile
rateprofile 1

set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 205
set roll_expo = 24
set pitch_expo = 24
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 22
set rate_center_sensitivity = 55
set rate_end_sensitivity = 125
set rate_center_correction = 25
set rate_end_correction = 75
set rate_center_weight = 50
set rate_end_weight = 40

# rateprofile
rateprofile 2

set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 210
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set rate_end_sensitivity = 125
set rate_end_correction = 75
set rate_end_weight = 40

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
'''

**Setup / Versions**
- NBD hummingbird Pro
- Betafpv Lite Radio 2 (supports Frsky D8, D16)
**Additional context**
D16 (FRSKY_X) working on NBD version of Betaflight 4.1.5 (betaflight_4.1.5_HMBF4_PRO_US.hex)
Tried EMU v0.3.3 and FRSKY_X setting - get rxloss at 30 feet or so indoors.  change to FRSKY_D (D8) and works as expected / normal range.   Indoor testing only in single family house.    Flashed back to NBD version betaflight_4.1.5_HMBF4_PRO_US.hex and D16 works as expected.

nerdCopter, 2021-01-04:
recommend D8 to circumvent the issue for the time being.
D8 gives me sticks + 4 switches without failsafes at least 200ft.

jgauruder, 2021-01-04:
Agreed on the D8 setting...that works for me just fine.

closed 2021-04-02 by Quick-Flash