emuflight/EmuFlight #460: RSSI channel not being saved - old issue, still present

szafranski, 2021-01-03:
**Describe the bug**
RSSI channel is not being saved (CLI and GUI - same thing).
There is some workaround (rssi_adc should be off in a first place), but it is not a proper fix I think.

**To Reproduce**
Steps to reproduce the behavior:
as in this video:
https://www.youtube.com/watch?v=x-qdaVpmn5o

**Expected behavior**
Expected:
RSSI channel should be saved but it is not.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
I won't post diff cause it is not related to any particulair FC.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.
It is old issue with a known workaround but it should not behave like this anyways - I think.

Quick-Flash, 2021-04-02:
yeah appears to be a gui bug rather than a firmware bug

nerdCopter, 2021-10-08:
i looked at the gui code for the old issue.

`src/js/tabs/receiver.js` line ~258 transfers the gui value into the variable `RSSI_CONFIG.channel = parseInt($('select[name="rssi_channel"]').val());`
then 
line ~267 saves with `MSP.send_message(MSPCodes.MSP_SET_RSSI_CONFIG, mspHelper.crunch(MSPCodes.MSP_SET_RSSI_CONFIG), false, save_rc_configs);`

so i dont know why it fails to save.

i set a debug line for console.log, and the save executes properly irregardless of `RSSI_ADC` enabled or disabled.

my conclusion is that the gui performs as expected.