emuflight/EmuFlight #463: Mapping beeper error on FF_RACEPIT: buzzer sound continuous.

Kamouloxfpv, 2021-01-05:
The problem exist with emuflight (check on 3.0.2 and 3.03),  on a FC **Furious FPV racepit**.
The buzzer solder on buz- and buz+ make continuous sound, when you plugg the lipo. impossible to stop even with radio.

This issue is cause by bad mapping in Emuflight. (no problem in betaflight).

you can found the error between mapping in betaflight and emuflight:
"resource BEEPER 1 B03" should be C03 in my case (just have to compare resource port in betaflight and emuflight difference)

**Betaflight:
# resource
resource BEEPER 1 C03**

**Emuflight
# resource
resource BEEPER 1 B03**

Betaflight (https://github.com/betaflight/betaflight/blob/master/src/main/target/FF_RACEPIT/target.h):

/*------------BEEPER---------------*/
#define USE_BEEPER
#define BEEPER_PIN              PC3

Emuflight (https://github.com/emuflight/EmuFlight/blob/master/src/main/target/FF_RACEPIT/target.h😞

/*------------BEEPER---------------*/
#define USE_BEEPER
#define BEEPER_PIN              PB3

solution is to **change the mapping** with good BEEPER_PIN in emuflight
in ex:
resource BEEPER 1 C03

THX to CC06 who solved this issue!

nerdCopter, 2021-05-25:
this issue had no activity, so i'm attempting to resolve myself.  would you mind testing this?
[EmuFlight_0.3.5_FF_RACEPIT_Build_a0d6296db.hex.zip](https://github.com/emuflight/EmuFlight/files/6539583/EmuFlight_0.3.5_FF_RACEPIT_Build_a0d6296db.hex.zip)

nerdCopter, 2021-07-14:
@ccafr06 :arrow_up:

Kamouloxfpv, 2021-07-20:
Hi nerdCopter, sorry for the delay, i dont have my racepit FC anymore, i cant test it :(
thank you for you work

nerdCopter, 2022-07-26:
#585

closed 2022-07-26 by nerdCopter