emuflight/EmuFlight #464: Target FF_RACEPIT - Wrong BEEPER PinIO assignment

ccafr06, 2021-01-05:
The beeper pin assignment for FC Racepit Furious FPV (target: FF-RACEPIT) is assign to pin B03 when it should be C03, causing the beeper to be ON constantly.

Betaflight settings (https://github.com/betaflight/betaflight/blob/master/src/main/target/FF_RACEPIT/target.h):
/*------------BEEPER---------------*/
#define USE_BEEPER
#define BEEPER_PIN              PC3

Emuflight (https://github.com/emuflight/EmuFlight/blob/master/src/main/target/FF_RACEPIT/target.h):
/*------------BEEPER---------------*/
#define USE_BEEPER
#define BEEPER_PIN              PB3

Setting it manualy in CLI issuing "resource BEEPER C03" solve the issue.

Extract of the conf:
# dump

# version
# EmuFlight / FF_RACEPIT (RACE) 0.3.2 Oct  7 2020 / 19:02:49 (e0270d5fd) MSP API: 1.49

board_name 
manufacturer_id 

# name
name Alien-Niko

# resources
**resource BEEPER 1 B03**
resource MOTOR 1 B11
resource MOTOR 2 B00
resource MOTOR 3 B10
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 C15
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 A08
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 C09
resource LED 1 B09
resource LED 2 B08
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 B00
resource CAMERA_CONTROL 1 A10
resource ADC_BATT 1 C02
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource PINIO 1 C00
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource SPI_PREINIT_IPU 1 A04
resource SPI_PREINIT_IPU 2 A15
resource SPI_PREINIT_IPU 3 B12
resource SPI_PREINIT_IPU 4 NONE
resource SPI_PREINIT_IPU 5 NONE
resource SPI_PREINIT_IPU 6 NONE
resource SPI_PREINIT_IPU 7 NONE
resource SPI_PREINIT_IPU 8 NONE
resource SPI_PREINIT_IPU 9 NONE
resource SPI_PREINIT_IPU 10 NONE
resource SPI_PREINIT_IPU 11 NONE
resource SPI_PREINIT_OPU 1 NONE
resource SPI_PREINIT_OPU 2 NONE

nerdCopter, 2022-05-02:
[EmuFlight_0.4.0b_FF_RACEPIT_Build_14e44c032.hex.zip](https://github.com/emuflight/EmuFlight/files/8603110/EmuFlight_0.4.0b_FF_RACEPIT_Build_14e44c032.hex.zip)
still need feedback

nerdCopter, 2022-07-26:
#585

closed 2022-07-26 by nerdCopter