emuflight/EmuFlight #468: Led menu cant change color and color 14 and 15 not working

qan313, 2021-01-13:
Cannot change colours in ledmenu with double clicking color. Im used to being able to change color 0 to 15. In betaflight when double clicking we get a menu to adjust the 3 color settings. For example 0,0,255. I can go from cli to set color 13 0,0,255. But in cli i also can't adjust the color in color 14 and color 15
Please assist

nerdCopter, 2021-01-13:
i'm unsure about the 14 and 15, but we do have a color-adjustment dialog box fix in a more recent build.
i can share an unofficial link for the moment.  it's identical to release except the color-adjustment dialog does not disappear upon mouse move.
https://drive.google.com/drive/folders/1GmXrou7lHGiuOWsqB4LqHaO3PGwsGqUD

qan313, 2021-01-13:
Thanks. This allready helps for setting color 0 to 13.
I will have to test trying to set color 14 and 15. 
I need those 2 for a more correct color spectrum when using potentiometer aux switch.

qan313, 2021-01-13:
Thanks, fixed

closed 2021-05-25 by nerdCopter