emuflight/EmuFlight #469: When setting up a new flightcontroller of the target DFRF722 the default settings include an enabled magnetometer and barometer. One of which is not supported by default and the other not supported if uart 3 is used.

PegasusFPV, 2021-01-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Kaiowarez, 2021-01-16:
> Assigned #469 <https://github.com/emuflight/EmuFlight/issues/469> to
> @Kaiowarez <https://github.com/Kaiowarez>.

Should I turn them off by default?

PegasusFPV, 2021-01-18:
Yes that would fix the issue for new pilots, and whoever wants to use a baro or magnetometer can turn it on.

Kaiowarez, 2021-01-18:
Ok cools will do that thanks for the advice

nerdCopter, 2021-10-21:
@Kaiowarez , would this modification fix both baro+mag or just baro (or neither?)
```diff
-#define SENSORS_SET             (SENSOR_ACC | SENSOR_BARO)
+#define SENSORS_SET             (SENSOR_ACC)
```

closed 2024-04-10 by nerdCopter