emuflight/EmuFlight #472: [Target Request] VGOODF722DUAL

nerdCopter, 2021-01-19:
requested by \@rampico

**Vendor/Model**
Please provide the target name. e.g. `VGOODF722DUAL`
Please provide vendor and model of the FC. e.g. `30.5x30.5mm V-Good / Racerstar StarF7 F7 Flight Controller AIO`
Please provide a lint to the vendor specifications.
https://www.banggood.com/30_5x30_5mm-Racerstar-StarF7-F7-Flight-Controller-AIO-OSD-5V-9V-BEC-Current-Sensor-Blackbox-and-40A-BLheli_32-2-6S-Brushless-ESC-for-DJI-Air-Unit-RC-Drone-FPV-Racing-p-1706211.html
https://www.racerstar.com/30_5x30_5mm-racerstar-starf7-f7-flight-controller-aio-osd-5v-9v-bec-current-sensor-blackbox-and-40a-blheli_32-2-6s-brushless-esc-for-dji-air-unit-rc-drone-fpv-racing-p-425.html

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/VGRC-VGOODF722DUAL.config
```
# 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: VGRC, board_name: VGOODF722DUAL, version: 5cc89407, date: 2020-06-24T00:54:00Z

# start the command batch
batch start

board_name VGOODF722DUAL
manufacturer_id VGRC

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
[... 1032 lines cut ...]
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
```

nerdCopter, 2021-06-16:
https://github.com/emuflight/EmuFlight/pull/594

nerdCopter, 2021-06-16:
@rampico, needs testing, no guarantees, please feedback.
[EmuFlight_0.3.4_VGRC_VGOODRCF722_DJI_Build_1330fb967.hex.zip](https://github.com/emuflight/EmuFlight/files/6665490/EmuFlight_0.3.4_VGRC_VGOODRCF722_DJI_Build_1330fb967.hex.zip)

rampico1, 2021-07-24:
Thanks nerdCopter!
Just uploaded the fw. Upload successfull. can get into the cli. Gyro not working. Motors not spinning. Amps/Voltage not working.RX on serial does work.

nerdCopter, 2021-07-26:
> Thanks nerdCopter!
> Just uploaded the fw.
> Upload successfull.
> at the boot only first 3 beeps.
> No osd.
> Can get into the cli.
> Gyro not working (drone in the UI is not moving)
> can't spin motors from the UI
> Amps/Voltage not working (in the UI)
> RX on serial does work!

@Kaiowarez :arrow_up:

rampico1, 2021-07-30:
@Kaiowarez 
cannot run further tests. Just fried the board after a crash.

BeauBrewski, 2021-11-14:
Needs tested!
Check gyro orientation, motor arrangement, flash, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY!

[EmuFlight_0.4.0_VGOODF722DUAL_Build_2d48add62.zip](https://github.com/emuflight/EmuFlight/files/7533645/EmuFlight_0.4.0_VGOODF722DUAL_Build_2d48add62.zip)

PR #1035 (2024-04-18): [target] add VGOODF722DUAL
resolves #472
closed 2024-04-18 by nerdCopter