emuflight/EmuFlight #473: [Target Request] MambaF405US_I2C

nerdCopter, 2021-01-19:
requested by \@swiss_fries

**Vendor/Model**
Please provide the target name. e.g. `MambaF405US_I2C`
Please provide vendor and model of the FC. e.g. `Diatone MAMBA F405 US I2C`
Please provide a lint to the vendor specifications.

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAF405US_I2C.config

BeauBrewski, 2021-06-09:
Needs tested:
https://discord.com/channels/547211754845765635/691022009865338950/814645999623733288

BeauBrewski, 2021-06-29:
Updated to 0.3.4, needs tested.

https://cdn.discordapp.com/attachments/691022009865338950/855183771769634816/EmuFlight_0.3.4_DIAT_MAMBAF405US_I2C_Build_1330fb967.hex

BadPixel89, 2021-09-12:
Hi, I just had a FC delivered that uses this target and I couldn't find it in the conifigurator. I would like to run 0.4.0 if possible, are there any steps I need to take?

nerdCopter, 2021-09-14:
> Hi, I just had a FC delivered that uses this target and I couldn't find it in the conifigurator. I would like to run 0.4.0 if possible, are there any steps I need to take?

[EmuFlight_0.4.0_DIAT_MAMBAF405US_I2C_Build_d07ad6e4f.hex.zip](https://github.com/emuflight/EmuFlight/files/7162121/EmuFlight_0.4.0_DIAT_MAMBAF405US_I2C_Build_d07ad6e4f.hex.zip)

nerdCopter, 2021-09-14:
#703

BadPixel89, 2021-09-14:
Flashes OK and connects to configurator.

Gyro and Accelerometer show readings below:

I moved the board on roll then pitch then yaw. Then followed with left-right translation forward-back translation and up-down translation.

[image]

 I noticed that barometer is greyed out despite the manual showing it has one and it being enabled in config tab:

[image]

[image]

Could this be because I was not plugging in a battery, I have not soldered anything to this stack. and I am testing using the FC alone (No ESC) plugged through USB.

nerdCopter, 2021-09-14:
i think some BMP280 fc's are needing manual code update.  this may be the case here as well.
we hope for better solution in time.
TYVM for testing!

nerdCopter, 2021-09-14:
i looked at code, it should support it as i found this:
```c++
#define USE_BARO
#define USE_BARO_MS5611
#define USE_BARO_BMP280
#define USE_BARO_BMP085
```
but maybe something else is amiss. possibly support for _external_ BARO.

BadPixel89, 2021-09-23:
Tested:
Receiver input working on the default pad
Cam / OSD working
Motors output as expected, test hovered OK

Not tested:
LED
Buzzer
SpeedyBee app, however this works on another Mamba stack with a different target, I just don't have the right cable.

BadPixel89, 2021-09-24:
Flight test today, all working as expected.

BadPixel89, 2021-10-01:
Speedybee does not work on this target, does work on the regular Mamba F405US. If anything else needs testing let me know.
Buzzer works.

nerdCopter, 2021-10-01:
@Kaiowarez :arrow_up:

BeauBrewski, 2021-11-12:
@nerdCopter I think these are the changes needed to fix the Barometer based off the Betaflight .config file. Just need to switch the I2C device to 1.

[image]

BadPixel89, 2022-01-27:
Update with Speedybee app. 
I realised it was the cable. It specifically needs to be an OTG compatible USBC cable. Which means the one I have for my laptop works, and many random USBC to USBA I've picked up work, but C-C doesn't seem to. 
I use an A-C adapter now, and that only works if the adapter is on the phone side, cable orientation doesn't seem to matter just directionality. 

In short USBC is weird with OTG, Speedybee is fine.

jbdu47, 2022-03-19:
Hy I use this target, the barometer is not recognized on emuflight.  would you have a solution?  thank you

nerdCopter, 2022-08-08:
Pretty sure this was fixed by #703  Sept 2021
Target : `DIAT_MAMBAF405US_I2C`

closed 2022-08-08 by nerdCopter

DragonBtc93, 2023-02-16:
I am having an odd issue with the osd. With camera unplug I can see the osd when the camera is plugged in the osd dissappears.

nerdCopter, 2023-02-16:
> I am having an odd issue with the osd. With camera unplug I can see the osd when the camera is plugged in the osd dissappears.

@DragonBtc93 , this is not a support-forum, please join Discord for support. the Short answer, is your Camera is probably set specifically to PAL or NTSC and your firmware is set to AUTO or the exact opposite setting of the camera.

DragonBtc93, 2023-02-16:
I tried that. Didn't work