emuflight/EmuFlight #479: [Target Request] Iflight Beast F7 45A AIO

AdrianLSY, 2021-01-30:
**Vendor/Model**
Please provide the target name:
IFLIGHT_F745_AIO

Please provide vendor and model of the FC:
Iflight Beast F7 45A AIO

Please provide a lint to the vendor specifications:
https://shop.iflight-rc.com/index.php?route=product/product&product_id=1377

**CLI dump or Unified Target**
Please provide a CLI `dump` or link to unified target:
https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO.config

nerdCopter, 2021-02-10:
@AdrianLSY , https://canary.discord.com/channels/547211754845765635/691022009865338950/808372714465263688

nene-ai, 2021-02-10:
Did someone have the Betaflight Firmware for this board?

nerdCopter, 2021-02-10:
@nene-ai https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO.config
@Kaiowarez , looks like it was modded 2 days ago, btw.

AdrianLSY, 2021-02-11:
> 
> 
> @AdrianLSY , https://canary.discord.com/channels/547211754845765635/691022009865338950/808372714465263688

@nerdCopter thanks, will test this out later

Mcrubens, 2021-02-16:
I hope this works. Target above had still some bugs. This is a fixed version: https://discord.com/channels/547211754845765635/691022009865338950/810629941980561459

Tested and seems to work fine except the LED_STRIP I think.

Mcrubens, 2021-02-16:
@nene-ai what would you need? I have a betaflight dump file if you want (see attached). Firmware you can flash through the configurator. 
[BTFL_cli_20210202_201728.txt](https://github.com/emuflight/EmuFlight/files/5991457/BTFL_cli_20210202_201728.txt)

nerdCopter, 2021-02-17:
@Kaiowarez , please see above comments. (view it on github).  Thanks!

AdrianLSY, 2021-02-27:
I tested the latest firmware, battery voltage and avg cell voltage showing 0.00v with a lipo plugged in

Mcrubens, 2021-02-27:
> I tested the latest firmware, battery voltage and avg cell voltage showing 0.00v with a lipo plugged in

In my case it worked. Used following settings: 

set current_meter = NONE
set battery_meter = ADC

but sometimes I have to retry. :-)

AdrianLSY, 2021-03-01:
> set battery_meter = ADC

yep, this fixed the issue

AdrianLSY, 2021-03-01:
> 
> 
> > I tested the latest firmware, battery voltage and avg cell voltage showing 0.00v with a lipo plugged in
> 
> In my case it worked. Used following settings:
> 
> set current_meter = NONE
> set battery_meter = ADC
> 
> but sometimes I have to retry. :-)

Yep this fixed the issue

gareth2263, 2021-03-30:
so just because i didnt follow any of that is there an emuflight target for the iflight beast f7 ?

nerdCopter, 2021-03-31:
@gareth2263 , yep, but it's a test-version that is only in Discord at the moment. check the link above if you are Discord.

nerdCopter, 2021-05-25:
[EmuFlight_0.3.4_IFRC_IFLIGHT_F745_AIO_Build_f4d0de6bf.hex.zip](https://github.com/emuflight/EmuFlight/files/6539485/EmuFlight_0.3.4_IFRC_IFLIGHT_F745_AIO_Build_f4d0de6bf.hex.zip)

frizzle101101, 2021-08-06:
can i ask permission to do this for inav using your code?

nerdCopter, 2021-08-08:
> can i ask permission to do this for inav using your code?

i don't quiet understand the question, but permission not required.  GPL, use and mod as needed.

nerdCopter, 2021-09-14:
`IFRC_IFLIGHT_F745_AIO` was merged awhile back for 0.4.0 release; however, there is minor issue with voltage: 
https://discord.com/channels/547211754845765635/691022009865338950/880876463967051848
```
Kaio Pryde: 
Target.c I added the dma
C08 0,1
E13 0,1
E9 0,2
E11 0,1
C7 0,1

Target.h needs this line
#define ADC_JNSTANCE ADC1 #define ADC1_DMA_STREAM DMA2_Stream4
#define USABLE_TIMER_CHANNEL_COUNT 15
```

BeauBrewski, 2022-04-01:
Fixed and merged. Closing.

closed 2022-04-01 by BeauBrewski