emuflight/EmuFlight #480: DJI Warnings hack does not work

Xedos9er, 2021-01-30:
**Describe the bug**
DJI Warning hack does not work

**To Reproduce**
Switch on "DJI warnings" in OSD Menu and place "Craft name" to osd. OSD in googles only show "Craft_name" instead of warnings.

**Expected behavior**
Warnings should be shown

**Flight controller configuration**
```
# version
# EmuFlight / HGLRCF722 (S7X2) 0.3.3 Jan 10 2021 / 18:57:34 (467b20241) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 00300039465350132030364e
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature
feature GPS

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 3 1 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 3 1300 1700 0 0

# adjrange

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtx

# rxfail

# master
set imuf_roll_q = 4000
set imuf_pitch_q = 4000
set imuf_w = 16
set acc_calibration = 16,0,-1
set min_check = 1030
set max_check = 1980
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 500
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 274
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 70
set gps_rescue_ground_speed = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = ON
set osd_warn_rc_smoothing = ON
set osd_warn_dji = ON
set osd_rssi_alarm = 35
set osd_alt_alarm = 150
set osd_vbat_pos = 2400
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2510
set osd_vtx_channel_pos = 0
set osd_crosshairs_pos = 206
set osd_current_pos = 2305
set osd_mah_drawn_pos = 2337
set osd_craft_name_pos = 2447
set osd_gps_speed_pos = 2241
set osd_gps_lon_pos = 2049
set osd_gps_lat_pos = 2065
set osd_gps_sats_pos = 2113
set osd_home_dir_pos = 2479
set osd_home_dist_pos = 2480
set osd_altitude_pos = 2272
set osd_warnings_pos = 331
set osd_avg_cell_voltage_pos = 2432
set osd_disarmed_pos = 2316
set debug_mode = GYRO_SCALED

# profile
profile 0

set dterm_lowpass_hz_roll = 110
set dterm_lowpass_hz_pitch = 110
set dterm_lowpass_hz_yaw = 110
set dterm_lowpass2_hz_roll = 180
set dterm_lowpass2_hz_pitch = 180
set dterm_lowpass2_hz_yaw = 180
set witchcraft_roll = 0
set witchcraft_pitch = 0
set spa_roll_p = 100
set spa_roll_i = 100
set spa_roll_d = 100
set spa_pitch_p = 100
set spa_pitch_i = 100
set spa_pitch_d = 100
set spa_yaw_p = 100
set spa_yaw_i = 100
set spa_yaw_d = 100
set iterm_rotation = OFF
set throttle_boost = 0
set feathered_pids = 0
set i_decay = 6
set emu_boost = 0
set emu_boost_yaw = 0
set emu_boost_limit = 0
set emu_boost_limit_yaw = 0
set p_pitch = 62
set i_pitch = 90
set p_roll = 57
set i_roll = 80
set d_roll = 26
set p_yaw = 75
set i_yaw = 75
set d_yaw = 20

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 25
set tpa_rate_p = 80
set tpa_rate_i = 120
```

**Setup / Versions**
-HGLRC Zeus F722
-Caddx Vista

nerdCopter, 2021-02-02:
@Xedos9er , i've asked around, but no replies. :(

Xedos9er, 2021-02-03:
@nerdCopter 
so you have the same issue? I know that the warnings work on INAV with the same FC. I tried it.
So it´s an issue in emoflight itself or the target.

nerdCopter, 2021-02-03:
@Xedos9er , i do not have DJI; i cannot test.  I asked others to test, but no response.
Does the DJI Hack work for you in the prior 0.3.2 release?  if yes, i can look at the source-code more deeply and search for changes that may have affected such.  i see to merges that possibly have affect, but only if 0.3.2 while 0.3.3 does not, then this might be true.  Please test 0.3.2 and reply with results. Thank you.

nerdCopter, 2021-02-03:
@nerdCopter , reference: https://github.com/emuflight/EmuFlight/pull/400/files , https://github.com/emuflight/EmuFlight/pull/389/files

nerdCopter, 2021-02-03:
@Xedos9er , do you have both craftname element and warnings element enabled on OSD?
i see this in your diff: `set osd_warnings_pos = 331`  when i use this on unsoldered test-FC, warnings are off-screen. (but i cannot personally test functionality)

try this config first:
```
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = ON
set osd_warn_rc_smoothing = ON
set osd_warn_dji = ON
set osd_craft_name_pos = 2346
set osd_warnings_pos = 2378
```

if still not working, please try this .hex and let me know if it acts differently:
[EmuFlight_0.3.3_HGLRCF722_Build_066feaa28.hex.zip](https://github.com/emuflight/EmuFlight/files/5919459/EmuFlight_0.3.3_HGLRCF722_Build_066feaa28.hex.zip)

Xedos9er, 2021-02-03:
Hi,
I tried your config, the hex and also 0.3.2, it just shows "CRAFT_NAME" in OSD, no warnings.
Then I flashed INAV and just activated craft name in OSD and set craft name as ":W", I see warnings in OSD.
It´s definately an issue with the target/Emuflight :(

Dutchcommando, 2021-02-03:
the warnings only come on when USB is disconnected. when connected to usb (configurator) the craft name shows in the "craft_name" field. this makes it easier to trouble shoot some stuff.

Xedos9er, 2021-02-03:
I know, that's why I tested without USB connected. There's some mess in the code.

nerdCopter, 2021-02-04:
additional info to note for records:
* User @Hasenberg reported worked on 0.3.3 NBD 20x20, but not matekF722se.

Xedos9er, 2021-02-04:
Then it has to be related to the target?
Since there's no documentation about the feature, is it enough to switch on the feature in OSD tab and just place the craft_name into the osd? 
In inav you need to name the craft ":W" to have warnings.

Dutchcommando, 2021-02-04:
for me it works on: 
-OmnibusF4SD
-betaflightF4
-HIFIONRCF7
-matek411RX
all running 0.3.3 after disconnecting usb and power cycle.

@Xedos9er you need to turn on (swap the switch to blue) _**DJI warnings**_ on the right hand side of the osd tab by warnings, it's default it off.
so: turn on **_CraftName_** and###  **DJI Warnings**, hit save and **_disconnect usb_**, plug in lipo (with props removed)

Xedos9er, 2021-02-05:
@Dutchcommando 
Exactly what I did: turn on CraftName and DJI Warnings. Then connect lipo and I only see "Craft_Name".
I also tried the CLI, but it´s the same as just switching on "DJI Warnings" in OSD Tab.

We can clearly say it´s a bug in Target :(

Xedos9er, 2021-02-05:
Some update:
I flashed today my 7" equipped with a KAKUTEHDV to emuflight 0.3.3. It works!
But I have to switch on the OSD Elements "Craft Name" and "Warnings" and toggle the DJI Warnings switch. Without switching on the warnings toggle it just shows "Craft_Name" in OSD.

On my HGLRCF722 it does not work, tried everything today, it must be an issue in the traget, 100% confirmed.

Xedos9er, 2021-02-08:
Yesterday I´ve done a flight and right after arming is saw "****" for a short period and "Craft_Name" disappeared for the whole flight. I tried to trigger some warnings by disarm, set throttle to 20% and try to arm, but I only heared the beeper and saw no warnings. 
Would be nice if we could solve this issue.

nerdCopter, 2021-02-08:
@mbilozub , any ideas on this stuff? i dont know how to solve or even what to look into. i dont have DJI, i dont think any of the active devs have DJI.  Thank you!

nerdCopter, 2021-02-25:
Question:  is there a difference between caddx-Vista vs DJI-Unit? or is it indeed FC target specific?

Xedos9er, 2021-02-26:
Yes, there´s no difference between vista and Air unit. That issue is very strange. I would also prefer the implementation from INAV, where you can choose between warnings and some additional information (Throttle, 3D speed, Trip distance).

RiskoFPV, 2021-02-26:
There is a pull request open that may fix this. #504 I was able to get it to work on TRANSTECF7, a previously non-working board.

nerdCopter, 2021-02-26:
Please test these and feedback working status, thank you.

[EmuFlight_0.3.3_FURYF4OSD_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6050019/EmuFlight_0.3.3_FURYF4OSD_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_BETAFPVF411_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6050020/EmuFlight_0.3.3_BETAFPVF411_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_MATEKF722_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6049994/EmuFlight_0.3.3_MATEKF722_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_TRANSTECF7_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6049995/EmuFlight_0.3.3_TRANSTECF7_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_MAMBAF722I2C_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6049993/EmuFlight_0.3.3_MAMBAF722I2C_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_HGLRCF722_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6049991/EmuFlight_0.3.3_HGLRCF722_Build_1374_a2fbd1126.hex.zip)
[EmuFlight_0.3.3_MAMBAF722_Build_1374_a2fbd1126.hex.zip](https://github.com/emuflight/EmuFlight/files/6049992/EmuFlight_0.3.3_MAMBAF722_Build_1374_a2fbd1126.hex.zip)

Xedos9er, 2021-02-27:
On HGLRCF722 it still not work.
On first power cycling i saw "BAT>FULL" message, then on arming "****", right after it stopped working. Sometimes it show "****" on arming, thats it. I tried to trigger RXFAIL an Throttle warning, but nothing appeared.

RiskoFPV, 2021-02-27:
When I was testing the warnings, at times it was a little laggy or unstable. After disarming it could take a few seconds to start updating again. I don't know if this is a problem with the fix, or because it is a workaround solution.

Dutchcommando, 2021-02-28:
i discovered that it has to do with either one specific OSD element, or the number of elements..
i had it working perfectly on a quad (hifonrc F7) but when i changed OSD (added a few elements, move some around) the warnings did not show anymore.. just craft name.. 
i did not have time to experiment, but once i find time i'll be trying to find out which element or what number makes the warnings stop from displaying.

Quick-Flash, 2021-04-02:
closing because this is fixed now.

closed 2021-04-02 by Quick-Flash