emuflight/EmuFlight #481: Add Arm status via Telemetry

Xedos9er, 2021-01-30:
**Is your feature request related to a problem? Please describe.**
In EMUflight you get no information about Arm status via Telemetry. Some lua scripts will not work, because they need arm status.

**Describe the solution you'd like**
Use same method as BF for arm status. Just add a "*" to flight mode when not armed.
Example:
ACRO* -> not armed
ACRO -> armed

**Describe alternatives you've considered**
no alternative, most scripts are meant to work with BF or INAV

Xedos9er, 2021-02-26:
Is there anyone who can implement this feature? Basically it can be copied 1:1 from Betaflight.

nerdCopter, 2021-02-26:
can you point to the BF Pull-Reqeust or "all" the code involved?

Xedos9er, 2021-02-26:
Since I'm not a coding guy, I can't tell you where you can find the part of the code involved into this.

Xedos9er, 2021-02-27:
@nerdCopter 
Tim Eckel created a pull request with all needed information:
https://github.com/emuflight/EmuFlight/pull/506

nerdCopter, 2021-02-27:
@Xedos9er what target FC's may i provide your your testing of # 506?

Xedos9er, 2021-02-27:
@nerdCopter 
of course i can test. Target HGLRCF722

nerdCopter, 2021-02-27:
[EmuFlight_0.3.3_HGLRCF722_Build_1385_4f23626d4.hex.zip](https://github.com/emuflight/EmuFlight/files/6055170/EmuFlight_0.3.3_HGLRCF722_Build_1385_4f23626d4.hex.zip)

Xedos9er, 2021-02-28:
@nerdCopter 
It´s working! Please implement into next version.

Quick-Flash, 2021-04-02:
this has been solved closing.

closed 2021-04-02 by Quick-Flash