emuflight/EmuFlight #486: BETAFPVF411RX Betafpv 1S AIO

dsrezoon, 2021-02-06:
**Vendor/Model**

BETAFPVF411RX
BETAFPV F4 1S AIO Brushless Flight Controller

https://betafpv.com/products/f4-1s-aio-brushless-flight-controller

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/tree/master/configs/default/BEFH-BETAFPVF411RX.config

BeauBrewski, 2021-07-20:
Seen where people on the Discord and FB page have used MatekF411Rx for this board with good results.

BeauBrewski, 2021-08-27:
Also seeing one case where unable to bind one of these boards with the PNP SPI receiver using MatekF411Rx

andreyk72, 2021-11-20:
No RX version of the board:
Gyro - OK, needed to calibrate accelerometer, by default it was like for "upside down" configuration. 
Accelerometer - OK, fly in angle mode no problem
Serial port 1 - OK (ELRS receiver)
Motors - OK
VTX control over Softserial port - OK
SPI RX (external via connector) - Didn't tested, do not use one
LEDS - not tested yet

BeauBrewski, 2021-11-20:
@andreyk72 awesome, thanks for testing and pointing out the alignment! The BetaFlight config has it as CW180, which is what I have it set at. The BetaFPV Meteor75 CLI from their website has it as CW0FLIP. So I think I should use that setting instead.

BeauBrewski, 2021-11-20:
@andreyk72 I made the change to the orientation and updated the file above *if* you want to recheck it. Thank you!

SKREAMfpv, 2021-11-22:
I can confirm that the BETAFPVF411RX target works with the external SPI receiver.

Thank you for working on this and getting it out @BeauBrewski!

BeauBrewski, 2021-11-22:
> I can confirm that the BETAFPVF411RX target works with the external SPI receiver.
> 
@SKREAMfpv That is *OUTSTANDING* news! Thank you for the feedback on that! Everything else good? Any noticeable issues?

SKREAMfpv, 2021-11-22:
@BeauBrewski Haven't experienced any issues with it yet! I was able to bind and LOS with this target in a babytooth setup in my living room. When the weather clears up I will maiden 'er FPV and report back!

nerdCopter, 2021-11-22:
> @andreyk72 awesome, thanks for testing and pointing out the alignment! The BetaFlight config has it as CW180, which is what I have it set at. The BetaFPV Meteor75 CLI from their website has it as CW0FLIP. So I think I should use that setting instead.

usually flightware should keep it at hardware's default, and if a vendor creates a quad that aligns it differently, such is not to be set as hardware default, but instead part of the pilot/model configuration.

Olafaloofian, 2022-01-13:
I have no OSD on my BetaFPV F4 AIO using this firmware

nerdCopter, 2022-01-13:
> I have no OSD on my BetaFPV F4 AIO using this firmware

even with OSD enabled?
[image]
or
```
feature OSD
save
```

BeauBrewski, 2022-03-11:
Target merged to master. CLosing

closed 2022-03-11 by BeauBrewski

tnttim1973, 2022-03-13:
> Seen where people on the Discord and FB page have used MatekF411Rx for this board with good results.

This fixed my issue. 

 I have the [F4 Brushed Flight Controller (SPI FUTABA RX + OSD)](https://betafpv.com/collections/brushed-flight-controller/products/f4-brushed-flight-controller-spi-futaba-rx-osd) . 

[image]

 When I updated to the latest firmware (BETAFPVF411RX 4.2.11) I lost communication to the receiver chips and the red LED would no longer light up. Only the Green LED stayed on constant. The bind button did nothing.  I tried older versions and still nothing. Right before I was about to throw it away, I seen this recommendation in the Matek version. In the software it does say that this is the chipset being used. 

 I tried the MATEKF411RX and as soon as it was completed the red LED came on and it was able to bind to my Futaba T8J. 

[image]

[image]

[image]

Here is the CLI Dump in Text. 

[BTFL_cli_20220313_013132.txt](https://github.com/emuflight/EmuFlight/files/8239071/BTFL_cli_20220313_013132.txt)

Hope this helps someone, I think this will be the last BetaFPV items I buy from them as they have very little to any support for their stuff. 
Tim

BeauBrewski, 2022-03-13:
@tnttim1973 
Betafpv support is a nightmare and confusing as heck. Just want to point out that on the link you sent, it does list `MATEKF411RX` as the firmware, as well as at the beginning of your BF dump.
[image]

And for future reference to make life easy, before flashing or upgrading, always check what target version came installed. It will be listed at the top left and bottom right of the configurator:
[image]

Or you can go into CLI and type `version`
[image]

Lastly, before pulling your hair out or tossing your FC in the trash, you can always ask on the Emuflight facebook page, discord, or on Github, and one of us will help you figure it out 😄 

Glad you got it figured out!