emuflight/EmuFlight #487: [Target Request] TMOTOR411 AIO

tom-bossa, 2021-02-10:
Target name : TMOTORF411

New T-Motor 13A F4 AIO that can be found in the Tron 80 Digital Whoop here : 
https://store-en.tmotor.com/goods.php?id=1136

Unified Target :
https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMOTORF411.config

❤🐥

tricade1, 2021-04-21:
@tom-bossa Can I use the target TMOTORF4 for the Tron 80HD ? Or do I have to wait until you added TMOTORF411? As I do have the Tron here and would love to test it!

nerdCopter, 2021-04-21:
i would have to assume that `F4` and `F411` are too different and would not work.

tricade1, 2021-04-21:
So the short info is - wait until you added it in emuflight :) correct ? Can’t wait !

tricade1, 2021-08-20:
I can see that in v0.4 STM32F411DISCOVERY has been added .... can I use this for the TRON? as its a STM32F411

nerdCopter, 2021-08-20:
> I can see that in v0.4 STM32F411DISCOVERY has been added .... can I use this for the TRON? as its a STM32F411

Please do not use the `STM*.hex` files at all, they are not real targets.  Emu is not unified targets.

BeauBrewski, 2021-11-15:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_TMTR_TMOTORF411_Build_3cfba270f.zip](https://github.com/emuflight/EmuFlight/files/7538427/EmuFlight_0.4.0_TMTR_TMOTORF411_Build_3cfba270f.zip)

tricade1, 2021-11-18:
very nice - does anybody has a good tune for EMU?

nerdCopter, 2021-11-18:
> very nice - does anybody has a good tune for EMU?

Every quad is different. need to tune or get "close" with the presets in the EmuConfigurator.  Join the EmuFlight Discord for help, Github is for development efforts and issue tracking.

BeauBrewski, 2022-03-10:
@tricade1 @tom-bossa 
Any feedback on this test target?
Any updates of functionality of this target?
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features, issues you noticed?

PR #1036 (2024-04-18): [target] add TMOTORF411
resolves #487
closed 2024-04-18 by nerdCopter