emuflight/EmuFlight #488: Clracing f7 dual gyro

NickBozos, 2021-02-14:
**Describe the bug**
Hello team. I have been using both clracing f7 dual gyro fcs v1 and v2 on my two freestyle quads and i can't make the blackbox to work. 

**To Reproduce**
Steps to reproduce the behavior:
I choose to the debug mode anything with "gyro" in it, as: gyro raw, dual gyro raw, gyro scaled,dual gyro diff. 

**Expected behavior**
Anything i choose and expecting to get gyro data, when i open the blackbox configurator, at the debug i get random results but never the debug mode i have chosen.
I tried to choose at the cli the wrong value i get for the debug, in order to get the one i really want in the blackbox but this didn't work either.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 clracing f7 dual gyro V1 and V2 ( banggood)
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 Emuflight 0.3- 0.3.3
 - how are the different components wired up.

**Additional context**
This happens in all versions of Emuflight i have tried . But it works fine when i tried to flash Betaflight just to check this issue.

NickBozos, 2021-03-01:
Are there any updates? Is something that i could try that would help?

nerdCopter, 2021-03-01:
@Kaiowarez +1 thing for you 10,000-item list :)

Kaiowarez, 2021-03-01:
Umm Blackbox working under normal gyro setting? Not debug I meant

NickBozos, 2021-03-01:
> Umm Blackbox working under normal gyro setting? Not debug I meant
> […](#)

Sorry, not sure i understood. Blackbox seems to work but i never get the debug mode i choose oin the cli.

NickBozos, 2021-03-17:
Is there any update?

nerdCopter, 2021-03-18:
No new information, sorry.  All i can assume is that it is probably due to dual-gyro.  Since EmuFlight is in process of re-basing on BF 4.X, i'm certain no effort will be put to address this in EmuFlight `0.X`.