emuflight/EmuFlight #508: [Target Request] FLYWOOF411V2

Roccccky, 2021-03-03:
**Vendor/Model**
GOKU GN F4 40A AIO BL_32 (MPU6000 ) 25.5x25.5
https://flywoo.net/products/goku-gnf4-40a-aio-bl_32-25-5x25-5

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411V2.config

**old version CLI dump or Unified Target (v1)**
https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411.config

**old version EMUFLIGHT Target (v1)**
https://github.com/emuflight/EmuFlight/tree/master/src/main/target/FLYWOOF411

BeauBrewski, 2021-11-15:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_FLYWOOF411V2_Build_50f6e92c0.zip](https://github.com/emuflight/EmuFlight/files/7540357/EmuFlight_0.4.0_FLYWOOF411V2_Build_50f6e92c0.zip)

BeauBrewski, 2022-03-02:
Merged, closing issue.
https://github.com/emuflight/EmuFlight/pull/749