emuflight/EmuFlight #510: Target Request - MATEKF722MINI

maxgame91, 2021-03-04:
Hi devs,

Is it possible to have a target for the matek F722 Mini
here on betaflight : https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/MTKS-MATEKF722MINI.config

https://www.drone-fpv-racer.com/matek-flight-controller-f722-mini-se-6817.html

Thanks :)

Mcrubens, 2021-03-04:
You can use to MATEKF722SE for that. You don't need the MINI. Works well except the LED_STRIP (but that is inherent as I guess the led drivers are not up to date).

maxgame91, 2021-03-04:
Thanks,
I will test that this week-end

iOperator, 2021-03-18:
The MATEKF722SE target works for my quad with a MATEKSYS F722-miniSE FC.

maxgame91, 2021-03-20:
Hi guys,

thanks guys, just flash the matekf722SE, but the baro is not detect

nerdCopter, 2021-03-22:
> Hi guys,
> 
> thanks guys, just flash the matekf722SE, but the baro is not detect

@Kaiowarez ? ^^^^^^^^^^^^^

Kaiowarez, 2021-03-22:
Another target I need to fix there's a few I think I forgot the drivers

iOperator, 2021-03-22:
The DPS310 baro was added rather recently to the Betaflight main branch if I remember correctly. Matek provided their own build to support it.
Let me see if I can create a pull request for the sensor.

goodnerdboy, 2021-04-29:
Hi
Any news on the DPS310 baro in emuflight?

closed 2026-05-10 by nerdCopter