emuflight/EmuFlight #517: [TARGET REQUEST] SKYSTARSF7HD

aepkolol, 2021-03-14:
**Vendor/Model**
SKYSTARSF7HD
SkyStars F7 3-6S 20x20 HD Flight Controller
https://www.getfpv.com/skystars-f7-3-6s-20x20-hd-flight-controller.html

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/SKST-SKYSTARSF7HD.config

BeauBrewski, 2021-07-14:
One of our pilots on the FB page said he used Diatone Mamba F722 on his SkystarF7HD and it worked. This is not the one recommended for this FC, but would not be the first time there's been rebadging and sharing. 

@aepkolol If you wanna try it at your own risk, check orientation, motors, etc (props off) and see if that gets you flying until an official build is made.

BeauBrewski, 2021-07-22:
Nothing official other than try MambaF722 at your own risk.

nerdCopter, 2021-07-28:
`July 28 [7:58 AM] @ bez: Can confirm mambaF722 image works [...]`

BeauBrewski, 2021-11-15:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY
[EmuFlight_0.4.0_SKYSTARSF7HD_Build_0412c109f.zip](https://github.com/emuflight/EmuFlight/files/7542248/EmuFlight_0.4.0_SKYSTARSF7HD_Build_0412c109f.zip)

byuddy7, 2021-11-29:
Hi there,
Thank you for putting this hex. I have the 30x30 version, but i think it is same chip as 20x20 as BF used same fw on both boards. Tested almost everything that I could think of, and so far I noticed issue only with the battery meter in my OSD is showing 0.0v. Previousily I used mambaf7 on this skystarf7hd board and no issue with battery meter. Attached dump and bb logs, hope this helps!
[SKYSTARSF7HD.zip](https://github.com/emuflight/EmuFlight/files/7618732/SKYSTARSF7HD.zip)

byuddy7, 2021-12-03:
@BeauBrewski Yes, I checked that both onboard ADC for Voltage and Amperage meter are turned on. Anyway I can see the voltage meter on my OSD and the configurator after I flashed my board with the new build hex file ver. ad8781124. I checked gyro orientation, motor arrangement, flash log, osd, and vtx control all are working. I will send it on this weekend and keep you posted if further issue are found. Many Thanks!

byuddy7, 2021-12-06:
@BeauBrewski flew it and the hex ver ad87811 works. So far no further issue. Thank you so much. Have a nice day, sir.

BeauBrewski, 2021-12-06:
@byuddy7 Thank you for your feedback! You rock!

BeauBrewski, 2022-03-11:
Merged to master. Closing.

BeauBrewski, 2022-08-14:
Two reports of motors 3 and 4 not working. Fixed by flashing `MAMBAF722`. 
OSD define needs fixed as well.

nerdCopter, 2024-04-18:
https://github.com/emuflight/EmuFlight/pull/770
https://github.com/emuflight/EmuFlight/pull/942

closed 2024-04-18 by nerdCopter