emuflight/EmuFlight #520: Tilt Effect does not work

iru-prog, 2021-03-17:
**Describe the bug**
After updating from 0.3.2 to 0.3.3, the flying feel of my tiny whoop has changed.
Specifically, in 0.3.3, I can't feel any change when I change the Tilt Effect value in Horizon mode.
In 0.3.2, I can feel the change. Am I doing something wrong? Please let me know.

**To Reproduce**
Change the Tilt Effect in Horizon mode.

**Expected behavior**
The force to return to horizontal is adjustable.

**Flight controller configuration**
# 

# diff all

# version
# EmuFlight / CRAZYBEEF4FR (C4FR) 0.3.3 Jan 10 2021 / 19:03:33 (467b20241) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 006a00553030510131333433
signature 

# reset configuration to default settings
defaults nosave

# name
name IRUFPV M6

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -AIRMODE
feature MOTOR_STOP

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 1 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 1700 2100 0 0
aux 5 35 2 1300 1700 0 0

# adjrange
adjrange 0 0 1 900 1200 0 1 0 0
adjrange 1 0 1 1200 1800 0 1 0 0
adjrange 2 0 1 1800 2100 0 1 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz_roll = 110
set gyro_lowpass_hz_pitch = 110
set gyro_lowpass_hz_yaw = 110
set imuf_roll_q = 7000
set imuf_pitch_q = 7000
set imuf_yaw_q = 7000
set imuf_w = 16
set imuf_sharpness = 3000
set acc_calibration = -65,-82,-37
set rssi_scale = 145
set rssi_offset = -35
set rc_interp = AUTO
set rc_interp_int = 19
set rc_smoothing_input_hz = 0
set rc_smoothing_derivative_hz = 0
set fpv_mix_degrees = 10
set rx_spi_protocol = FRSKY_D
set dshot_idle_value = 1000
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 10
set bat_capacity = 300
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 32
set vbat_scale = 114
set ibata_scale = 1075
set ibatv_scale = 1175
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set deadband = 1
set yaw_deadband = 1
set runaway_takeoff_prevention = OFF
set osd_cap_alarm = 300
set osd_vbat_pos = 12
set osd_rssi_pos = 2401
set osd_tim_2_pos = 375
set osd_flymode_pos = 2457
set osd_throttle_pos = 2392
set osd_vtx_channel_pos = 2421
set osd_current_pos = 388
set osd_mah_drawn_pos = 2432
set osd_craft_name_pos = 2442
set osd_pidrate_profile_pos = 2369
set osd_warnings_pos = 2376
set osd_avg_cell_voltage_pos = 2412
set osd_disarmed_pos = 266
set osd_core_temp_pos = 2405
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 4
set vtx_freq = 5800
set vcd_video_system = NTSC
set frsky_spi_tx_id = 54,223
set frsky_spi_offset = -8
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0

# profile
profile 0

set dterm_lowpass_hz_roll = 150
set dterm_lowpass_hz_pitch = 150
set dterm_lowpass_hz_yaw = 150
set dterm_lowpass2_hz_roll = 250
set dterm_lowpass2_hz_pitch = 250
set dterm_lowpass2_hz_yaw = 250
set witchcraft_roll = 0
set witchcraft_pitch = 0
set spa_roll_p = 100
set spa_roll_i = 100
set spa_roll_d = 100
set spa_pitch_p = 100
set spa_pitch_i = 100
set spa_pitch_d = 100
set spa_yaw_p = 100
set spa_yaw_i = 100
set spa_yaw_d = 100
set iterm_windup = 50
set feathered_pids = 0
set i_decay = 3
set emu_boost = 400
set emu_boost_yaw = 400
set emu_boost_limit = 100
set emu_boost_limit_yaw = 100
set p_pitch = 74
set i_pitch = 75
set d_pitch = 38
set p_roll = 80
set i_roll = 75
set d_roll = 38
set p_yaw = 80
set i_yaw = 75
set d_yaw = 0
set horizon_tilt_effect = 0

# profile
profile 1

set dterm_lowpass_hz_roll = 100
set dterm_lowpass_hz_pitch = 100
set dterm_lowpass_hz_yaw = 100
set dterm_lowpass2_hz_roll = 250
set dterm_lowpass2_hz_pitch = 250
set dterm_lowpass2_hz_yaw = 250
set iterm_windup = 50
set p_pitch = 80
set i_pitch = 25
set d_pitch = 80
set p_roll = 80
set i_roll = 25
set d_roll = 80
set p_yaw = 90
set i_yaw = 90
set d_yaw = 0
set horizon_tilt_effect = 130

# profile
profile 2

set dterm_lowpass_hz_roll = 100
set dterm_lowpass_hz_pitch = 100
set dterm_lowpass_hz_yaw = 100
set dterm_lowpass2_hz_roll = 250
set dterm_lowpass2_hz_pitch = 250
set dterm_lowpass2_hz_yaw = 250
set iterm_rotation = OFF
set iterm_windup = 50
set feathered_pids = 30
set emu_boost = 0
set emu_boost_yaw = 0
set emu_boost_limit = 0
set emu_boost_limit_yaw = 0
set horizon_tilt_effect = 130

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set tpa_rate_p = 60
set tpa_rate_i = 100
set tpa_rate_d = 70

# rateprofile
rateprofile 1

set tpa_breakpoint = 1200

# rateprofile
rateprofile 2

set tpa_breakpoint = 1200

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
**Setup / Versions**
CRAZYBEEF4FR

Quick-Flash, 2021-04-02:
the tilt effect doesn't change the strength to get back to center. It rather changes the angle at which point you are fully in acro mode.

iru-prog, 2021-04-02:
Thank you for your advice. However, since 0.3.3, I can't fly with the same feeling as 0.3.2 because the behavior doesn't change when I change the value. Do you have any hints for me?

closed 2022-07-26 by nerdCopter