emuflight/EmuFlight #522: TMOTOR PACER F7

legionfpv, 2021-03-22:
**TMotor/Pacer F7**
Please provide the target name. e.g. `TMOTORPACERF7`
Please provide vendor and model of the FC. e.g. `TMotor Pacer F7 F722 Flight Controller`
Please provide a lint to the vendor specifications.
https://store-en.tmotor.com/goods.php?id=1086
board_name TMPACERF7
manufacturer_id TMTR

# name: -

# resources
resource BEEPER 1 C14
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
[... 1001 lines cut ...]
# rateprofile 0
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

nerdCopter, 2021-03-23:
There is a dev/test version in Discord (still needs testing, afaik):
https://canary.discord.com/channels/547211754845765635/691022009865338950/819927844137730080

brunocannavina, 2021-04-26:
Yes please, have the same FC and I want to use emuflight like my other quads! :)
I can't open the discord link.

legionfpv, 2021-04-26:
Cant access it also, using betaflight Tmotor/PacerF7 target, until Emu gets
it on board.
Emuflight is slowing down could this mean people are shifting elsewhere?

Thanks

nerdCopter, 2021-04-26:
@brunocannavina , @legionfpv 
Sorry about the link problem.  Go into Emu Discord and search `from: Kaio Pryde#6694 has: file` you should find it.

EmuFlight devs still very busy/active, btw.

legionfpv, 2021-04-26:
Thanks nerdCopter, will look into it later.
Glad you are all still keen on this great flight tool!!!

brunocannavina, 2021-04-28:
On 3/9/2021 I bought a drone FT5 MK2 60%.
At that time I did not have a receiver, so I tried it connected to the computer and everything seemed to work perfectly.

Last Friday I received the receiver and, when trying to arm with the radio, I noticed that the motors made a strange noise.
so I plugged it back into the computer and tried to arm again, but the noise was gone.

I think the problem is that the 5V BEC built into the FC is emitting a lot of noise,
particularly in the gyro (pitch).

I have tried everything, updating the firmware, tightening the screws, activating the rpm filter, etc.

I just have to try emuflight, because I have a lot of faith in it.

Can you share with me your settings?
Thanks.

BeauBrewski, 2021-06-09:
The most up-to-date version of this firmware. Needs tested.

https://discord.com/channels/547211754845765635/691022009865338950/819927844137730080

commit febd629c (2021-06-21): tmtr (#600)

closes #522 #455
closed 2021-06-21 by nerdCopter