emuflight/EmuFlight #525: [Target Request] HGLRC Zeus25 AIO 3-6S F722 Flight Controller 25A BL32 4in1 ESC

r7972963, 2021-03-26:
**Vendor/Model**
HGLRC Zeus25 AIO 3-6S F722 Flight Controller 25A BL32 4in1 ESC
https://www.hglrc.com/products/hglrc-zeus25-aio-3-6s-f722-flight-controller-25a-bl32-4in1-esc

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-ZEUSF722_AIO.config

aepkolol, 2021-04-28:
Came here to request this FC as well.

adracea, 2021-06-30:
Would also like to request this target.

nerdCopter, 2021-06-30:
@Kaiowarez , is this "`HGLR_HGLRCF722`" ??

nerdCopter, 2021-06-30:
[EmuFlight_0.3.4_HGLR_HGLRCF722_Build_1330fb967.hex.zip](https://github.com/emuflight/EmuFlight/files/6743087/EmuFlight_0.3.4_HGLR_HGLRCF722_Build_1330fb967.hex.zip)

please test and feedback, TY!

adracea, 2021-06-30:
> @Kaiowarez , is this "`HGLR_HGLRCF722`" ??

The only difference in the configs seems to be the ZEUS25 also having these lines changed, I'm not sure how this affects things though.  
```
set gyro_1_sensor_align = CW180FLIP
set gyro_1_align_pitch = 1800
```

Kaiowarez, 2021-07-01:
Might just need a sub branch to make those changes

BeauBrewski, 2021-07-02:
> @Kaiowarez , is this "`HGLR_HGLRCF722`" ??

Should be HGLR-ZEUSF722_AIO according to manufacturer website and BF config.

BeauBrewski, 2021-11-14:
Needs tested!
Check gyro orientation, motor arrangement, flash, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_HGLR-ZEUSF722_AIO_Build_4b10ba7ba.zip](https://github.com/emuflight/EmuFlight/files/7534145/EmuFlight_0.4.0_HGLR-ZEUSF722_AIO_Build_4b10ba7ba.zip)

nerdCopter, 2021-11-14:
@BeauBrewski please rename your target without hyphen. use underscore only, ty.

RustyRobotbits, 2022-02-27:
First flight was successful. Used this .hex for my Zeus35 pro aio fc. Will continue to test.

BeauBrewski, 2022-03-10:
@RustyRobotbits 
Any feedback is much appreciated!
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features, issues you noticed?

PR #835 (2022-07-29): TARGET ZEUS722_AIO
Tested per https://github.com/emuflight/EmuFlight/issues/525
Closes #525
closed 2022-07-29 by nerdCopter