emuflight/EmuFlight #526: iflight h7 55a

bspires123, 2021-03-27:
**Vendor/Model**
Please provide the target name. e.g.IFLIGHT_H743_AIO
Please provide vendor and model of the FC. e.g. Iflight Beast H7 55A AIO (MPU6000)
Please provide a lint to the vendor specifications.  https://shop.iflight-rc.com/index.php?route=product/product&path=20_27_330&product_id=1376
**CLI dump or Unified Target**
Please provide a CLI `dump` or link to unified target. https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO.config
  Find the unified targets here: https://github.com/betaflight/unified-targets/tree/master/configs/default

bspires123, 2021-03-27:
sorry newbee!!!

nerdCopter, 2021-03-27:
@bspires123 , sorry i did not notice this was for `H7`.  H7 support will not exist in 0.3.X, 0.4.X. 
it will only be supported in 1.0.0, thank you.

closed 2021-03-27 by nerdCopter

bspires123, 2021-03-27:
I don't understand?
so there is no support for this?
can i try emuflight on this f/c/
please help

nerdCopter, 2021-03-28:
no support for H7 chips yet.  sorry.
must wait for EmuFlight 1.0.0 which is in progress, but far from release.

bspires123, 2021-03-28:
ok thank you