emuflight/EmuFlight #532: Issue with VGOODF4 or VGOODF4DJI

nerdCopter, 2021-04-09:
**Describe the bug**
Issue with VGOODRCF4 (or maybe unreleased VGOODRCF405_DJI)

@Diversionpfe commented on orignal PR #410:
>
> Accept my apologies if this is not the right place to post this issue.
> 
> My 3in cinewhoop is based on Eachine's Tyro79 stack which has the above FC together with a 4-in-1 ESC that supports 20A constant current/25A peak current, ONESHOT 125/42, DSHOT 150/300/600, based on MCU EFM8BB21F16G-C.
>I flashed the FC with the latest EmuFlight VGRC/VGOODRC and found out that my gyro has a 90 degrees CW displacement (YAW is ok but PITCH acts like ROLL and ROLL acts like PITCH when I hold the drone in my hand and I am looking the drawing in SETUP screen).
> Is this normal? Why can't the default values make the FC be 0 degrees oriented?
> The only way I discovered to solve this is using CW 90degrees but the overall response looks a bit non linear when compared to VGRC/VGOODRCF4 firmware flashing in BF without any orientation displacement.
> Also, and this happens regardeless I flash the 4 ESCs either with BLHeli or BlueJay, I have only one working motor. Physically is number 2 but it appears as Motor 1 in Motor Screen.
> Am I missing something in my config?
> Thanks in advance

@Zlopi posted the original PR for VGOODRCF4.

@Diversionpfe has tested released VGOODRCF4 as well as builds from @Kaiowarez 's unified-branch:
VGRC_VGOODRCF4 had similar issue, but failing motors are different.
VGRC_VGOODRCF405_DJI had no gyro, no accel.

Attached are @Diversionpfe's dumps: (but i dont think they are defaults)
[DIversionpfe_BetaFlight_VGOODRCF4_dump_all_2021apr8.txt](https://github.com/emuflight/EmuFlight/files/6285925/DIversionpfe_BetaFlight_VGOODRCF4_dump_all_2021apr8.txt)
[DIversionpfe_Emuflight_dump_all_2021apr8.txt](https://github.com/emuflight/EmuFlight/files/6285926/DIversionpfe_Emuflight_dump_all_2021apr8.txt)

o0ZzM4g1c, 2021-04-18:
Hi there. I have a Tyro79 with a VGOODRCF4 target if you need some tests or information on issue (had same issue as Diversionpfe

nerdCopter, 2021-06-16:
https://github.com/emuflight/EmuFlight/pull/594

nerdCopter, 2021-06-16:
@o0ZzM4g1c , @Diversionpfe , i guess we need some testing. no clue if these are good or not.  built from PR.
[VGoodF_F4s_need_testingAndFeedback.zip](https://github.com/emuflight/EmuFlight/files/6665326/VGoodF_F4s_need_testingAndFeedback.zip)

Diversionpfe, 2021-06-16:
Hi there!
I will flash it tomorrow.
Thkx!

nerdCopter, 2021-06-21:
* @Diversionpfe reported `VGOODRCF4` needed `CW270`.
* @Diversionpfe reported `VGOODRCF405_DJI` not seeing gyro on `MINI405DJIPAD` hardware.

o0ZzM4g1c, 2021-06-21:
> 
> 
> @o0ZzM4g1c , @Diversionpfe , i guess we need some testing. no clue if these are good or not. built from PR.
> [VGoodF_F4s_need_testingAndFeedback.zip](https://github.com/emuflight/EmuFlight/files/6665326/VGoodF_F4s_need_testingAndFeedback.zip)

I need some repair work on motor, then I'll flash it.  Which kind of test do you need ? / validate ?

nerdCopter, 2021-06-21:
Yessir, make sure everything works as expected.  gyro, accel, orientation, motors, bbl flash, LED if applicable, ...

o0ZzM4g1c, 2021-09-03:
Hello, I upgraded FW to 0.4.0 VGOODRCF4. then I still have motor resource issue. what can I do ?

nerdCopter, 2021-09-07:
> Hello, I upgraded FW to 0.4.0 VGOODRCF4. then I still have motor resource issue. what can I do ?

please work with @Kaiowarez for fixing