emuflight/EmuFlight #533: [Target Request] BHHD/EMAXF411(STM32F411)

dtmaupin, 2021-04-12:
**Vendor/Model**
* Please provide the target name. e.g. `BHHD/EMAXF411(STM32F411)`
Vendor and model of the FC: 'EMAX BabyHawk II HD BLHeli_S F4 2-4S 25A AIO FC + ESC`
Link to the vendor specifications.
https://emax-usa.com/collections/babyhawk-ii/products/babyhawk-ii-hd-part-b-aio-flight-controller-esc-board

**CLI Dumps from EMAX support pages with both configurations**
 
[CLI_Babyhawk_II_HD_v1.2.txt](https://github.com/emuflight/EmuFlight/files/6293421/CLI_Babyhawk_II_HD_v1.2.txt)
[CLI_Babyhawk_II_HD_v1.3（TBS Crossfire RX).txt](https://github.com/emuflight/EmuFlight/files/6293422/CLI_Babyhawk_II_HD_v1.3.TBS.Crossfire.RX.txt)

BeauBrewski, 2021-11-15:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_EMAXF411_Build_8d3212e19.zip](https://github.com/emuflight/EmuFlight/files/7537989/EmuFlight_0.4.0_EMAXF411_Build_8d3212e19.zip)

BeauBrewski, 2022-03-10:
Thoroughly tested and fixed BLHELI interface/motors.

BeauBrewski, 2022-03-11:
Target merged. Closing.

closed 2022-03-11 by BeauBrewski