emuflight/EmuFlight #539: Unable to arm drone due to inaccurate initial voltage reading causing 1S LiPo to register as 2S

schome1, 2021-04-14:
**Describe the bug**
When I plug in my fully charged 1S LiHV battery, the voltage in the Power State section of the Power & Battery tab spikes to 4.6V.  Since my Maximum Cell Voltage was set to 4.4V, the firmware set the cell count to 2S.  As a result, when trying to arm the drone, I was unable to because a 2S battery reading 4.35V is way too low to arm.  

Note that the voltage does drop to the correct reading.  It's the initial voltage reading that spikes to an erroneous voltage, which is also used to calculate the cell count.

A temporary workaround is to set the Maximum Cell Voltage to 4.7 (above the max voltage that is read initially) so the cell count will be calculated correctly, despite the inaccurate voltage reading.

**To Reproduce**
1. Charge a 1S LiHV to full charge (4.35V)
2. Plug-in your drone and connect to EmuFlight Configurator
3. Go to the Power & Battery tab
4. Set your Maximum Cell Voltage to 4.4 or 4.5
5. plug in the fully charged 1S LiHV to your drone that is plugged into USB and being configured via EmuFlight
6. Notice the Voltage spike to an unrealistic value in the Power State section of the Power & Battery tab, then notice that the Connected value is equal to Yes (Cells: 2) instead of Yes (Cells: 1).  
7. Redo steps 4 through 6, only change the Maximum Cell Voltage to 4.7 or higher.  Note that when you plug in a fully charged 1S LiHV, now the Connected values will return Yes (Cells: 1)

**Expected behavior**
A reasonable voltage should be able to be entered as a Maximum Cell Voltage parameter value (something like 4.4 or 4.5 for a 1S LiHV for the cell count to be accurate.

**Flight controller configuration**
```
# version
# EmuFlight / HMBF4PRO_US (HMBF4 PRO) 0.3.3 Jan 10 2021 / 19:01:35 (467b20241) MSP API: 1.50

board_name 
manufacturer_id 

# name
name Schome-Wan

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature MOTOR_STOP

# beeper

# beacon

# map

# serial
serial 0 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1175 1350 0 0
aux 1 1 1 1900 2100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 28 0 1225 1275 0 0
aux 4 35 2 1300 1350 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set vbat_max_cell_voltage = 47
set vbat_scale = 112
set ibata_scale = 1659
set yaw_motors_reversed = ON
set osd_vbat_pos = 2391
set osd_rssi_pos = 2368
set osd_tim_2_pos = 2412
set osd_flymode_pos = 2106
set osd_vtx_channel_pos = 2400
set osd_current_pos = 2423
set osd_mah_drawn_pos = 2380
set osd_craft_name_pos = 2090
set vtx_freq = 5917
set vcd_video_system = NTSC
set frsky_spi_tx_id = 45,76
set frsky_spi_offset = -13
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0

# profile
profile 0

# rateprofile
rateprofile 2

set roll_rc_rate = 200
set pitch_rc_rate = 150
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
```

**Setup / Versions**
 - NewBeeDrone Hummingbird F4 Pro, purchased from NewBeeDrone
 - NewBeeDrone NitroNectar Gold 300mAh LiHV, NewBeeDrone BDR Gold Brushless 18000kv 0802 motors
 - Motors are the only things soldered to the FC.  Everything else is either on-board or plugged in.

**Additional context**
- Note that after the initial voltage reading, it drops down to actual voltage in the display.

Quick-Flash, 2021-04-14:
you should still be able to arm with very low voltages. Just type this in the cli

set force_battery_cell_count = 1
save

that will make it always think that its a 1s.

nerdCopter, 2022-08-08:
thinking this could be calibrated as well
[image]

closing ticket.

closed 2022-08-08 by nerdCopter